esphome/esphome #10612: Warning! Could not find the package with 'Wire @ 2.0.0' requirements for your system 'linux_aarch64'

bwinklesky, 2025-09-05:
### The problem

When attempting to install the Wire library, the following error message is displayed.

`Warning! Could not find the package with 'Wire @ 2.0.0' requirements for your system 'linux_aarch64'` 

If you try using Wire in a component, it fails because it can't find Wire.h.  

### Which version of ESPHome has the issue?

2025.8.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

Custom I2C

### YAML Config

```yaml
esphome:
  name: test
  friendly_name: test
  libraries:
    # a library bundled with Arduino
    - Wire
esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

### Anything in the logs that might be useful for us?

```txt
Running HA on a VirtualBox VM.
```

bwinklesky, 2025-09-06:
For a component library that is using Wire.

swoboda1337, 2025-09-06:
Wire is part of Arduino and you are using the esp-idf framework.

bwinklesky, 2025-09-06:
I'm not sure what that has to do with this issue.  If I build a ESP32 firmware using PlatformIO and esp-idf I can use the Wire without any problems.

swoboda1337, 2025-09-06:
Wire is a built in library which is part of Arduino: https://github.com/espressif/arduino-esp32/tree/master/libraries/Wire. You need to use the Arduino framework or use Arduino as a component (ESPHome does not support yet).

swoboda1337, 2025-09-06:
> Like I said, it works if I use the esp-idf framework using PlatformIO. And I thought ESPHome uses PlatformIO. Maybe it's just not possible with your python scripts.

Again Wire is part of Arduino, you need to use Arduino is some way. Either the arduino framework or as an idf component.

bwinklesky, 2025-09-06:
I understand it's part of Arduino.  But if you import a library that isn't compatible, why don't you mention that it's not compatible when it builds?  platformIO it seems to handle it fine using idf.

bwinklesky, 2025-09-06:
Thank you for the aggressive insights.  Not even sure why I was asked why I'm trying to use Wire.  Seems irrelevant

ssieb, 2025-09-06:
If you're making a component for esphome, you should be using the esphome components instead of trying to use Wire directly.

ssieb, 2025-09-06:
> Not even sure why I was asked why I'm trying to use Wire. Seems irrelevant

We would rather help you not need it than help you do something that's not really supported.

bwinklesky, 2025-09-06:
Thanks for the help! I appreciate it

closed 2025-09-06 by swoboda1337

m0uh85, 2025-09-07:
I have the same issue.... until ESPHome 6.3 everything was fine and compiling without any issue. But now I get the same error like described at the top.

[logs_info-e-paper-display_run.txt](https://github.com/user-attachments/files/22195731/logs_info-e-paper-display_run.txt)

swoboda1337, 2025-09-07:
> I have the same issue.... until ESPHome 6.3 everything was fine and compiling without any issue. But now I get the same error like described at the top.
> 
> Can someone please try to help me to fix the issue?
> 

That is not the same issue as you are actually using Arduino. Can you file your own issue as this one has been closed.