esphome/esphome #10625: EZO - I2C read error since 2025.08.02

HpNoTiQ56, 2025-09-06:
### The problem

Since 4 days, both ezo sensor (ORP+PH) can't read data. I tried to rollback to 2025.8.0 ESPHOME and rebuild but still can't get data.

`[esp-idf:000]: E (41591426) i2c.master: I2C software timeout
[23:08:11][D][esp-idf:000]: E (41591430) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[23:08:11][D][esp-idf:000]: E (41591432) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[23:08:11][E][ezo.sensor:088]: read error`

### Which version of ESPHome has the issue?

ESPHOME 2028.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

I2C

### YAML Config

```yaml
substitutions:
  device_name: esp4-piscine
  friendly_name: esp4
  adress_ip: "XXXXX"
  time_timezone: "Europe/Paris"

#external_components:
#  - source: github://nrandell/dallasng

esphome:
  name: ${device_name}

esp32:
  board: lolin_d32
  framework:
    type: esp-idf
#    version: recommended

wifi:
  networks:
#    - ssid: !secret wifi_shelly
#      password: !secret mdpwifi_shelly
#      priority: 0    
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
      priority: 1
      hidden: true
    
  
  reboot_timeout: 5min
  fast_connect: true

  manual_ip:
    static_ip: ${adress_ip}
    gateway: XXX
    subnet: XXX

# Enable logging
logger:

# Enable bluetooth tracker/proxy

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: false

bluetooth_proxy:
  active: true

# Enable Home Assistant API
api:
  encryption:
    key: "XXX"

ota:
  - platform: esphome
# Web server à désactiver si proxy bluetooth actif (out of memory)
# web_server:
#   port: 80

# Example configuration entry for ESP32
i2c:
  sda: 21
  scl: 22
  scan: false
  id: bus_a
  frequency: 300kHz

#Etat de la connection
binary_sensor:
  - platform: status
    name: "${friendly_name}_Status"

# Sonde Dallas
one_wire:
  - platform: gpio
    pin: GPIO32

#dallasng:
#  - pin: 32
#    update_interval: 3s

button:
  - platform: template
    name: "Clear PH Calibration" 
    id: ph_calibrate_clear
    on_press:
      then: 
        - lambda: id(esp4_ph_ezo).clear_calibration();
  - platform: template
    name: "PH 6.86 Calibrate" 
    id: ph_calibrate_6_86
    on_press:
      then: 
        - lambda: id(esp4_ph_ezo).set_calibration_point_mid(6.86);
  - platform: template
    name: "PH 9.18 Calibrate" 
    id: ph_calibrate_9_18
    on_press:
      then: 
        - lambda: id(esp4_ph_ezo).set_calibration_point_high(9.18);
  - platform: template
    name: "PH 4 Calibrate" 
    id: ph_calibrate_4
    on_press:
      then: 
        - lambda: id(esp4_ph_ezo).set_calibration_point_low(4.00);
  - platform: template
    name: "Export" 
    id: Export
    on_press:
      then: 
        - lambda: id(esp4_ph_ezo).send_custom("Slope,?");

sensor:

# Mesure du pH
  - platform: ezo
    id: esp4_ph_ezo
    name: ph_ezo
    address: 99
    unit_of_measurement: "pH"
    accuracy_decimals: 2
    update_interval: 60s
  #  moyenne sur 15 mn-affichage toutes les 5mn
  # Etalonné en 2023
    filters:
 #     - calibrate_linear:
 #       - 4.44 -> 4.0
 #       - 7.17 -> 6.86
 #       - 9.41 -> 9.18
       - sliding_window_moving_average:
          window_size: 15
          send_every: 5
          send_first_at: 1

# Mesure de l'ORP
# Etalonné le 13 mai 2025 / sonde Meytec Or
  - platform: ezo
    id: esp4_orp_ezo
    name: orp_ezo
    address: 98
    unit_of_measurement: "mV"
    accuracy_decimals: 0
    update_interval: 60s
    filters:
#      - multiply: 10
      - calibrate_linear:
          - 0 -> 0
          - 580 -> 645
 # moyenne sur 15 mn-affichage toutes les 5mn
      - sliding_window_moving_average:
          window_size: 15
          send_every: 5
          send_first_at: 1

  # Estimation du taux de chlore
  - platform: template
    name: "Chlore estimé (ppm)"
    unit_of_measurement: "ppm"
    accuracy_decimals: 2
    update_interval: 60s
    icon: mdi:chemical-weapon 
    lambda: |-
      const float orp = id(esp4_orp_ezo).state;  // en mV
      const float ph = id(esp4_ph_ezo).state;    // pH de 0 à 14

      if (orp <= 0 || ph <= 0) {
        return NAN;
      }

      // Formule empirique basée sur un modèle simplifié
      // Source : adaptation de modèles de traitement d'eau et docs industrielles
      float ppm = 10 * exp((orp - 650.0) / 100.0) * pow(10, -(ph - 7.0));

      // Optionnel : borne les valeurs
      if (ppm < 0.0) ppm = 0.0;
      if (ppm > 5.0) ppm = 5.0;

      return ppm;

# Mesure de la pression filtre
  - platform: adc
    pin: A0
    id: esp4_pression_filtre
    name: "Pression filtre"
    unit_of_measurement: "Bars"
    update_interval: 60s
    attenuation: 12db
    filters:
      - calibrate_linear:
        - 0.58 -> 0.0
        - 0.82 -> 0.8
# moyenne sur 30 mn + affichage toutes les mn
      - sliding_window_moving_average:
          window_size: 30
          send_every: 1
# Avec le 30PSI
#    filters:
#      - calibrate_linear:
#        - 0.5 -> 0.0
#""        - 2.21 -> 0.8
# moyenne sur 30 mn + affichage toutes les mn
#      - sliding_window_moving_average:
#          window_size: 30
#          send_every: 1

  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal Sensor"
    update_interval: 60s

#  - platform: homeassistant
#    name: "T° Eau Piscine"
#    id: temp_piscine
#    entity_id: sensor.temp_piscine

  - platform: dallas_temp
    name: "T° Eau Piscine"
    id: temp_piscine
    address: 0x0c236c761f64ff28
    accuracy_decimals: 1
    update_interval: 5s
  
  # - platform: dallasng
  #  name: "T° Eau Piscine"
  #  address: 0x0c236c761f64ff28
  #  id: temp_piscine

# gestion afficheur
display:
  - platform: lcd_pcf8574
    dimensions: 16x2
    address: 0x27
    update_interval: 5s
    lambda: |-
      it.printf(0,0,"Ph=%.2f",id(esp4_ph_ezo).state);
      it.printf(0,1,"Or=%.1f",id(esp4_orp_ezo).state);
      it.printf(9,0,"P=%.3f",id(esp4_pression_filtre).state);
      it.printf(9,1,"T=%.1f",id(temp_piscine).state);

switch:    
  - platform: restart
    name: "${friendly_name}_Restart"
    

text_sensor:
  # IP address #
  - platform: wifi_info
    ip_address:
      name: "${friendly_name} IP address"
      icon: "mdi:network-outline"
    ssid:
      name: "${friendly_name} Connected SSID"
    bssid:
      name: "${friendly_name} Connected BSSID"
  - platform: version
    id: text_sensor_version
    name: "${friendly_name} ESPHome Version"
```

### Anything in the logs that might be useful for us?

```txt
[esp-idf:000]: E (41591426) i2c.master: I2C software timeout
[23:08:11][D][esp-idf:000]: E (41591430) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[23:08:11][D][esp-idf:000]: E (41591432) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[23:08:11][E][ezo.sensor:088]: read error
```

swoboda1337, 2025-09-07:
> I've tried all version from 2025.8

Can you try with 2025.8.3 and this:
```
external_components:
  - source:
      type: git
      url: https://github.com/swoboda1337/esphome
      ref: i2c_test
    refresh: always
    components: [ i2c ]
```

HpNoTiQ56, 2025-09-07:
Still same error, maybe timeout is not long enough, PH or ORP sensor was long.

> [15:49:06][D][esp-idf:000]: E (65038) i2c.master: I2C software timeout
[15:49:06][D][esp-idf:000]: E (65042) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[15:49:06][D][esp-idf:000]: E (65044) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[15:49:06][E][ezo.sensor:088]: read error

mortification77, 2025-09-09:
There is no real fix for now it seems besides using an older ESPHome Version.  You can do this WITHOUT a full ESPHome downgrade and use an older version to compile specific devices only.

I tried the alternate i2c Packages with the same result.

For a QUICK AND EASY solution for HomeAssistant:

Add via HACS:  https://github.com/khenderick/esphome-legacy-addons

Install the version you want:  2025.7 (Worked to Resolve I2C Issues in 2025.8*)

Then, go into Settings -> Add-ons, and you will see the additional ESPHome version listed.  Click it, START IT, and then click "Open WebUI".  Use THIS INTERFACE to compile problematic devices, and all will be well!

swoboda1337, 2025-09-10:
Looking at the datasheet I think the EZO component has issues:

[image]

The max delay is 900ms not 1s and it reads 32 bytes not 7.

mortification77, 2025-09-10:
The issues that I have encountered are generic I2C issues, hence my comment...

I have had multiple issues with I2C devices (NACK and Transaction Failed).  The issue appears to be in the esp-idf framework's I2C in general since nothing I could do in the I2C settings or component made any difference at all.  I spent WAY too much time thinking it was a possible device issue.

1)  0.96 Display on a TTGO LoRa MCU.  I gave up and set this one aside for the time being.

2)  I have a device with a QMC5883L Magnetometer that is used to pick up on magnetic changes as my water meter spins.  It worked and was quite stable for nearly a year until 2025.8, at which point the device was throwing the I2C Errors and began to restart every 2-10 minutes.   After using 2025.7 yesterday, it has now been up for 15 HOURS.

I just noticed a notification that 2025.8.4 was released and an i2c fix was implemented:

[kmeteriso] Fix i2c call [esphome#10618](https://github.com/esphome/esphome/pull/10618) by [@clydebarrow](https://github.com/clydebarrow)

Fingers crossed!

mortification77, 2025-09-10:
> That 2025.8.4 fix was to a single component.
> 
> The IDF I2C driver was rewritten and it is revealing some issues with a few components that just happened to work with the old driver. For example the EZO component the delay is supposed to be 1000ms according to the datasheet but it is only 900ms. It is possible there are issues with driver itself though.

There are certainly issues with the driver.  I can see issues with a couple of random devices coming up when a driver is "fixed", but not this many...

swoboda1337, 2025-09-10:
> One irritating thing is that the I2C errors are marked as DEBUG level. Events at this level should _NOT_ lead to system instability / restarts.

Do you have a log?

mortification77, 2025-09-10:
From the screenshots I did this morning from my mobile:

[09:15:41][D][esp-idf:000]: E (9839)  i2c.master: I2C hardware NACK detected
[09:16:18][D][esp-idf:000]: E (31281) i2c.master: I2C hardware NACK detected

I have a feeling that the "E" in the message after the "message header" means "ERROR", but I'm not positive.  Maybe the ESPHome Logger is not treating them as it should?

swoboda1337, 2025-09-20:
@HpNoTiQ56 did you try the increased timeouts?
```
external_components:
  - source:
      type: git
      url: https://github.com/swoboda1337/esphome
      ref: i2c_test
    refresh: always
    components: [ i2c, ezo ]
```