esphome/esphome #10633: Warning! Could not find the package with 'Wire @ 2.0.0' requirements for your system 'linux_aarch64'

m0uh85, 2025-09-07:
### The problem

I try to update my screen but since v6.3 I get the Warning msg

[logs_info-e-paper-display_run.txt](https://github.com/user-attachments/files/22196394/logs_info-e-paper-display_run.txt)

### Which version of ESPHome has the issue?

>2025.6.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml
substitutions:
  run_time: 5min #can be as long as needed to get data 
  sleep_time: 20min # normal sleep time
  night_sleep_time: 6h # 1st sleep time after midnight
  
esphome:
  name: "info-e-paper-display"
  friendly_name: Info - E-Paper Display
  platformio_options:
    # Include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus
    # Directory where the lilygo-t5-47-plus.json file can be found
    board_dir: boards
  on_boot:
    priority: -100
    then:
      - pcf8563.read_time
  # custom build path, so we can save the lilygo-t5-47-plus.json file in the sub-folder "boards"
  build_path: /config/esphome/e-paper/

  libraries:
    - SPI
    
esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus

  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API (API password is deprecated in favor of encryption key)
# https://esphome.io/components/api.html
api:
  encryption:
    key: "1NYteWxgVKd6CR1cbmwRF+rFK+CmByTh79I8dNWH9Cc="

web_server: # creates a web server where you can access all this stuff without home assistant (good for debugging or working headless (no HA))
  port: 80
  include_internal: true

ota:
  platform: esphome
  password: "130ca244f05998d2f666d34848bbbed2"

wifi:
  ssid: !secret wifi_ssid
[... 549 lines cut ...]
    }

    // Energie Verbrauch
    y = 320;
    it.printf(powerCol, y, id(font_icons_small), TextAlign::BASELINE_RIGHT, ICON_solar); 
    if (!isnan(id(solar_power).state)) {
      it.printf(xPower, y, id(font_large), TextAlign::BASELINE_RIGHT, "%.1f", id(solar_power).state);
      it.printf(xPower, y, id(font_medium), TextAlign::BASELINE_LEFT, "W");
    }
    y += rowHeight;
    it.printf(powerCol, y, id(font_icons_small), TextAlign::BASELINE_RIGHT, ICON_energy); 
    if (!isnan(id(power_consumption).state)) {
      it.printf(xPower, y, id(font_large), TextAlign::BASELINE_RIGHT, "%.1f", id(power_consumption).state);
      it.printf(xPower, y, id(font_medium), TextAlign::BASELINE_LEFT, "W");
    }

    // ** Vertical lines
    it.line(xCenter + 105, 270, xCenter + 105, yRes);
```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-09-07:
The problem is the external component you are using: https://github.com/nickolay/esphome-lilygo-t547plus/blob/main/components/t547/display.py#L52

It is trying to use version 2.0.0 of the Wire library but the current version is 3.2.1

In the ESPHome code we dont specify a version https://github.com/esphome/esphome/blob/dev/esphome/components/i2c/__init__.py#L96 however that external component broke things.

swoboda1337, 2025-09-07:
If you add wire to libraries:
```
  libraries:
    - SPI
    - Wire@3.2.1
```
it will fail with:
```
ValueError: Version pinning failed! Libraries Wire@2.0.0 and Wire@3.2.1 requested with conflicting versions!
```
You will have to fill an issue in the github of the external component