esphome/esphome #10667: I2C and MCP9808 not booting with latest Esphome

jkaberg, 2025-09-10:
### The problem

As per title, affects Esphome 2025.8.X AFAIK - worked fine on 7.X. See logs an config sections:

### Which version of ESPHome has the issue?

2025.8.X

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

I2C MCP9808

### YAML Config

```yaml
esphome:
  name: varmtvannsbereder
  friendly_name: varmtvannsbereder

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: github://pr#10618
    components: [kmeteriso]
    refresh: 1h

# Enable logging
logger: 
  level: debug

# Enable Home Assistant API
api:
  encryption:
    key: !secret encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 10.0.40.19
    gateway: 10.0.40.1
    subnet: 255.255.255.0
    dns1: 10.0.40.1

  output_power: 8.5dB

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Varmtvannsbereder"
    password: !secret ap_fallback_password

bluetooth_proxy:
  active: true

button:
  - platform: restart
    name: "Restart"

i2c:
  sda: 22 #green
  scl: 23 #white
  frequency: 400kHz

sensor:
#  - platform: mcp9808
#    name: "VVB Top Temperature"
#    address: 0x18
#    filters:
#      - min:
#          window_size: 3
#          send_every: 1

#  - platform: mcp9808
#    name: "VVB Center Temperature"
#    address: 0x19
#    filters:
#      - min:
#          window_size: 3
#          send_every: 1

#  - platform: mcp9808
#    name: "VVB Lower Temperature"
#    address: 0x1A
#    filters:
#      - min:
#          window_size: 3
#          send_every: 1

  - platform: dht
    pin: 25
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
bomming
```

jkaberg, 2025-09-10:
Sorry if I'm unnecessarily pinging you @clydebarrow but I saw you did some work on the I2C component in recent commits

leblancexplores, 2025-09-13:
I'm having odd behavior too with my sensors after upgrading from 8.2 to 8.4. My BME680 (22/23) I think fried itself pulling 6W continuous (smelt it) and I'm still trying to isolate the problem and why now my LD2410C (16/17) is pulling 1W at boot and only about 8/10 times shows sensor data in Home Assistant.

I thought this might be because I installed a LED/PWM control via pin 19 that is ran through a Si4599 (N type Mosfet side) to a pulled up (to 12V) ND1 (w/ 10K resistor), this part of course works flawlessly.

Sorry for ths long winded post but this is a i2C device too so thought I'd ask, I'm going to try to downgrade to 8.4.2, I have another BME680 on order but this might get expensive if this thing is the culprit. Anyone know if I am missing something? I'm just running inverted 3.3 -> 0 on the NS1 (GND) and NG1 (19) pins to pull the drain pin which goes to a [Automotive PWM fan controller ](https://a.co/d/8k47Csv)

swoboda1337, 2025-10-07:
> and it works as before. So if ESP messed up something in I2C driver and that version contains a fix, the 5.5.1 should definitely be released ASAP in the mainstream.

There is a PR for it https://github.com/esphome/esphome/pull/9839. However there was a platform issue and an Arduino compile issue. We are supposed to get a new release of those tomorrow. Its too late for 2025.10 but 5.5.1 will be in 2025.11. You will also be able to use 5.5.1 in 2025.10 like you did above.

Liniik, 2025-10-07:
Well, october is still going so I don't really understand why is it a problem to put it into a 2025.10.X release but whatever. I'm on dev and working so that's okay I guess...

closed 2026-01-07 by jkaberg