esphome/esphome #10706: ESPHome will pass raw data to Home Assistant ignoring precision and rounding specifications in the filter.

andrewjswan, 2025-09-14:
### The problem

ESPHome will pass raw data to Home Assistant ignoring precision and rounding specifications in the filter.
This results in an increase in the size of the database and an unnecessary increase in long term statistics.

### Which version of ESPHome has the issue?

2024.8.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

api

### YAML Config

```yaml
platform: total_daily_energy
name: Energy
icon: mdi:lightning-bolt
power_id: power
unit_of_measurement: kWh
device_class: energy
state_class: total_increasing
method: ${energy_method}
accuracy_decimals: 3
filters:
  - multiply: 0.001
  - sliding_window_moving_average:
      window_size: 20
      send_every: 20
  - round: 3
web_server:
  sorting_weight: 25
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

[image]

andrewjswan, 2025-09-14:
> What do the device logs show?

Strange update time, although it is specified in the filter `send_every: 20`, for some reason it sends more often:

```
[13:41:30][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:31][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:32][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:33][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:34][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:35][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:36][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:37][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:38][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
[13:41:39][D][sensor:103]: 'Energy': Sending state 0.00000 kWh with 3 decimals of accuracy
```

andrewjswan, 2025-09-15:
> You don't show what the source is

```
platform: template
name: Power
icon: mdi:flash
id: power
lambda: |-
  return id(current).state * 5; // A * 5V
unit_of_measurement: W
device_class: power
state_class: measurement
accuracy_decimals: 1
update_interval: 3s
disabled_by_default: true
filters:
  - round: 1
web_server:
  sorting_weight: 15
```

> `send_every` means samples, not time.

Yes, I know, the source is updated every 3 seconds (`update_interval: 3s`), every 20 should update energy `once a minute`.

> It's sending 0, what are you seeing in HA?

Same thing, **update every second**. Both in the interface and in the database, although the value does not change (in the interface).

[image]

andrewjswan, 2025-09-15:
> Did you filter that log?

Yes

> Why aren't there any log messages from the current and power sensors updating?

They came in packs, the values ​​were constant at that moment. They did not change, just like Energy. If necessary, I will give a log with them.

andrewjswan, 2025-09-15:
[image]

closed 2025-10-01 by swoboda1337

ssieb, 2025-12-31:
And it's still not an esphome issue.

andrewjswan, 2025-12-31:
> And it's still not an esphome issue.

It's complex here, if you look at this message https://github.com/esphome/esphome/issues/10706#issuecomment-3289436794 It is evident that the sensor refresh rate is not as specified, but much more frequent. Moreover, this issue only occurs with ESPHome devices.

andrewjswan, 2025-12-31:
> Create a new issue with more info if you still have an update interval problem.

I'll open it, no problem, but what other data is missing? All the data is there, including the configuration and log.