esphome/esphome #10738: I2C Hardware NACK and continuous boot loop on wt32-eth01 and ESPHome 2025.8.3

deirdreobyrne, 2025-09-15:
### The problem

Getting an i2c hardware NACK with version 2025.8.3, which then seems to precipitate a reboot. Problem doesn't exist with 2025.7.5

### Which version of ESPHome has the issue?

2025.8.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

i2c

### YAML Config

```yaml
esphome:
  name: atticnw
  friendly_name: atticnw

esp32:
  board: wt32-eth01
  framework:
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

text_sensor:
  - platform: version
    name: "ESPHome Version"

button:
  - platform: restart
    name: "Restart"

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    duration: 33s
    active: true

bluetooth_proxy:
  active: true

web_server:
  port: 80
  local: True

i2c:
  sda: GPIO33
  scl: GPIO32
  id: i2cbus

sensor:
  - platform: bme280_i2c
    pressure:
      name: "Pressure"
    address: 0x76
  - platform: htu21d
    model: si7021
    temperature:
      name: "Attic Temperature"
    humidity:
      name: "Attic Humidity"
    heater:
      name: "Attic sensor heater"
```

### Anything in the logs that might be useful for us?

```txt
[C][component:164]: Setup ethernet took 1935ms
[VV][app:427]: logger loop enabled from ISR
[V][ethernet:454][sys_evt]: [Ethernet event] ETH started (num=0)
[V][ethernet:454][sys_evt]: [Ethernet event] ETH connected (num=2)
[D][esp32_ble:287]: Enabling
[D][esp32_ble_tracker:219]: Starting scan, set scanner state to STARTING.
[V][esp32_ble_client:044]: [0] [] Set state 2
[V][esp32_ble_client:044]: [1] [] Set state 2
[V][esp32_ble_client:044]: [2] [] Set state 2
[I][ethernet:273]: Starting connection
[W][component:289]: ethernet set Warning flag: waiting for IP configuration
[V][ethernet:519]: DHCP Client Status: 1
[V][i2c.idf:130]: Writing 1 bytes, reading 0 bytes
[V][i2c.idf:141]: 0x40 TX F3
[V][i2c.idf:172]: Sending 4 jobs
[VV][component:216]: logger loop disabled
[V][esp32_ble:344]: gap_event_handler - 2
[V][esp32_ble_tracker:330]: gap_scan_set_param_complete - status 0
[V][esp32_ble:344]: gap_event_handler - 7
[V][esp32_ble_tracker:340]: gap_scan_start_complete - status 0
[V][esp32_ble:320]: gattc_event [esp_gatt_if: 3] - 0
[V][esp32_ble_client:266]: [0] [] gattc_event_handler: event=0 gattc_if=3
[V][esp32_ble_client:272]: [0] [] gattc registered app id 1
[V][esp32_ble:320]: gattc_event [esp_gatt_if: 4] - 0
[V][esp32_ble_client:266]: [1] [] gattc_event_handler: event=0 gattc_if=4
[V][esp32_ble_client:272]: [1] [] gattc registered app id 2
[V][esp32_ble:320]: gattc_event [esp_gatt_if: 5] - 0
[V][esp32_ble_client:266]: [2] [] gattc_event_handler: event=0 gattc_if=5
[V][esp32_ble_client:272]: [2] [] gattc registered app id 3
[V][esp32_ble_tracker:306]: gap_scan_result - event 0
[VV][component:216]: bluetooth_proxy loop disabled
[VV][component:216]: bluetooth_proxy loop disabled
[VV][component:216]: bluetooth_proxy loop disabled
[V][i2c.idf:130]: Writing 0 bytes, reading 2 bytes
[V][i2c.idf:153]: 0x40 RX bytes 2
[V][i2c.idf:172]: Sending 5 jobs
[D][htu21d:056]: Got Temperature=23.5ТАC
[V][sensor:051]: 'Attic Temperature': Received new state 23.549545
[D][sensor:103]: 'Attic Temperature': Sending state 23.54955 ТАC with 1 decimals of accuracy
[V][i2c.idf:130]: Writing 1 bytes, reading 0 bytes
[V][i2c.idf:141]: 0x40 TX F5
[V][i2c.idf:172]: Sending 4 jobs
[W][component:423]: htu21d.sensor took a long time for an operation (52 ms)
[W][component:424]: Components should block for at most 30 ms
[V][i2c.idf:130]: Writing 0 bytes, reading 2 bytes
[V][i2c.idf:153]: 0x40 RX bytes 2
[V][i2c.idf:172]: Sending 5 jobs
[D][htu21d:077]: Got Humidity=52.4%
[V][sensor:051]: 'Attic Humidity': Received new state 52.395386
[D][sensor:103]: 'Attic Humidity': Sending state 52.39539 % with 1 decimals of accuracy
[V][i2c.idf:130]: Writing 1 bytes, reading 2 bytes
[V][i2c.idf:141]: 0x40 TX 11
[V][i2c.idf:153]: 0x40 RX bytes 2
[V][i2c.idf:172]: Sending 8 jobs
[VV][esp-idf:000]: E (6177) i2c.master: I2C hardware NACK detected
```

deirdreobyrne, 2025-09-19:
Nope :(  Same result as before.

swoboda1337, 2025-09-19:
> Nope :( Same result as before.

Does it still say `[VV][esp-idf:000]: E (6177) i2c.master: I2C hardware NACK detected` or does it say something else?

swoboda1337, 2025-09-19:
Can you try again there were some serious issues:
```
external_components:
  - source: github://pr#10801
    components: [htu21d]
    refresh: always
```

deirdreobyrne, 2025-09-19:
> Can you also confirm you do have a SI7021 and not a HTU21D or SHT21?

Yes it's an SI7021 - it was all working very well up until I tried to install 2025.8.3

swoboda1337, 2025-09-19:
I am surprised 2025.7 works as the component has a few buffer overruns like this: 
```
  int8_t raw_heater;
  if (this->read_register(HTU21D_READHEATER_REG_CMD, reinterpret_cast<uint8_t *>(&raw_heater), 2) != i2c::ERROR_OK) {
```

swoboda1337, 2025-09-19:
Did the heater sensor actually work previously?