esphome/esphome #10740: display.mipi_dsi & LVGL on M5Stack Tab5 rotation & color issues

RAR, 2025-09-15:
### The problem

I'm attempting to setup a M5Stack Tab5 to use LVGL in landscape and getting a blank screen (backlight is on).  It does work in portrait mode, however the colors are off from the expected defaults when including LVGL with no options.

You can use the example on https://devices.esphome.io/devices/M5Stack-Tab5 and just modify the display to add a rotation and include LVGL:

```
display:
  - platform: mipi_dsi
    rotation: 90
    id: my_display
    model: M5Stack-Tab5
    reset_pin:
      pi4ioe5v6408: pi4ioe1
      number: 4

lvgl:
```

In the logs I noticed that while display.mipi_dsi does note the width/height change, LVGL does not:

```[13:03:03][C][display.mipi_dsi:347]: MIPI_DSI RGB LCD
[13:03:03][C][display.mipi_dsi:347]:   Model: M5STACK-TAB5
[13:03:03][C][display.mipi_dsi:347]:   Width: 1280
[13:03:03][C][display.mipi_dsi:347]:   Height: 720
[13:03:03][C][display.mipi_dsi:347]:   Mirror X: NO
[13:03:03][C][display.mipi_dsi:347]:   Mirror Y: NO
[13:03:03][C][display.mipi_dsi:347]:   Swap X/Y: NO
[13:03:03][C][display.mipi_dsi:347]:   Rotation: 0 degrees
[13:03:03][C][display.mipi_dsi:347]:   DSI Lanes: 2
[13:03:03][C][display.mipi_dsi:347]:   Lane Bit Rate: 730Mbps
[13:03:03][C][display.mipi_dsi:347]:   HSync Pulse Width: 40
[13:03:03][C][display.mipi_dsi:347]:   HSync Back Porch: 140
[13:03:03][C][display.mipi_dsi:347]:   HSync Front Porch: 40
[13:03:03][C][display.mipi_dsi:347]:   VSync Pulse Width: 4
[13:03:03][C][display.mipi_dsi:347]:   VSync Back Porch: 20
[13:03:03][C][display.mipi_dsi:347]:   VSync Front Porch: 20
[13:03:03][C][display.mipi_dsi:347]:   Buffer Color Depth: 16 bit
[13:03:03][C][display.mipi_dsi:347]:   Display Pixel Mode: 16 bit
[13:03:03][C][display.mipi_dsi:347]:   Color Order: RGB
[13:03:03][C][display.mipi_dsi:347]:   Invert Colors: NO
[13:03:03][C][display.mipi_dsi:347]:   Pixel Clock: 60MHz
[13:03:03][C][display.mipi_dsi:375]:   Reset Pin 4 via PI4IOE5V6408
[13:03:03][C][lvgl:088]: LVGL:
[13:03:03][C][lvgl:088]:   Display width/height: 720 x 1280
[13:03:03][C][lvgl:088]:   Buffer size: 100%
[13:03:03][C][lvgl:088]:   Rotation: 90
[13:03:03][C][lvgl:088]:   Draw rounding: 2
```

To take LVGL out of the equation I added a page under display and it DID show in landscape with the correct color:

```
color:
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%

display:
  - platform: mipi_dsi
    rotation: 90
    id: my_display
    model: M5Stack-Tab5
    reset_pin:
      pi4ioe5v6408: pi4ioe1
      number: 4
    pages:
      - id: page1
        lambda: |-
          it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
          it.rectangle(0, 20, it.get_width(), it.get_height(), id(my_blue)); 
          it.rectangle(0, 100, it.get_width(), it.get_height(), id(my_blue));  
```

### Which version of ESPHome has the issue?

2025.8.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

mipi_dsi

### YAML Config

```yaml
esphome:
  name: display_deva

esp32:
  board: esp32-p4-evboard
  flash_size: 16MB
  framework:
    type: esp-idf
    advanced:
      enable_idf_experimental_features: true

esp32_hosted:
  variant: esp32c6
  active_high: true
  clk_pin: GPIO12
  cmd_pin: GPIO13
  d0_pin: GPIO11
  d1_pin: GPIO10
  d2_pin: GPIO9
  d3_pin: GPIO8
  reset_pin: GPIO15
  slot: 1

psram:
  mode: hex
  speed: 200MHz

logger:
  level: DEBUG
  hardware_uart: USB_SERIAL_JTAG

api:
  encryption:
    key: "encryption!"

ota:
  - platform: esphome
    password: "password!"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

captive_portal:

i2c:
  - id: bsp_bus
    sda: GPIO31
    scl: GPIO32
    frequency: 400kHz

spi:
  id: spi_bus
  clk_pin: GPIO5
  mosi_pin: GPIO18
  miso_pin: GPIO19

pi4ioe5v6408:
  - id: pi4ioe1
    address: 0x43
    # 0: O - wifi_antenna_int_ext
    # 1: O - speaker_enable
    # 2: O - external_5v_power
    # 3: NC
    # 4: O - lcd reset
    # 5: O - touch panel reset
    # 6: O - camera reset
    # 7: I - headphone detect
  - id: pi4ioe2
    address: 0x44
    # 0: O - wifi_power
    # 1: NC
    # 2: NC
    # 3: O - usb_5v_power
    # 4: O - poweroff pulse
    # 5: O - quick charge enable (inverted)
    # 6: I - charging status
    # 7: O - charge enable

switch:
  - platform: gpio
    id: wifi_power
    name: "WiFi Power"
    pin:
      pi4ioe5v6408: pi4ioe2
      number: 0
    restore_mode: ALWAYS_ON
  - platform: gpio
    id: usb_5v_power
    name: "USB Power"
    pin:
      pi4ioe5v6408: pi4ioe2
      number: 3
  - platform: gpio
    id: quick_charge
    name: "Quick Charge"
    pin:
      pi4ioe5v6408: pi4ioe2
      number: 5
      inverted: true
  - platform: gpio
    id: charge_enable
    name: "Charge Enable"
    pin:
      pi4ioe5v6408: pi4ioe2
      number: 7

  - platform: gpio
    id: wifi_antenna_int_ext
    pin:
      pi4ioe5v6408: pi4ioe1
      number: 0
  - platform: gpio
    id: speaker_enable
    name: "Speaker Enable"
    pin:
      pi4ioe5v6408: pi4ioe1
      number: 1
    restore_mode: ALWAYS_ON
  - platform: gpio
    id: external_5v_power
    name: "External 5V Power"
    pin:
      pi4ioe5v6408: pi4ioe1
      number: 2

binary_sensor:
  - platform: gpio
    id: charging
    name: "Charging Status"
    pin:
      pi4ioe5v6408: pi4ioe2
      number: 6
      mode: INPUT_PULLDOWN

  - platform: gpio
    id: headphone_detect
    name: "Headphone Detect"
    pin:
      pi4ioe5v6408: pi4ioe1
      number: 7

select:
  - platform: template
    id: wifi_antenna_select
    name: "WiFi Antenna"
    options:
      - "Internal"
      - "External"
    optimistic: true
    on_value:
      - if:
          condition:
            lambda: return i == 0;
          then:
            - switch.turn_off: wifi_antenna_int_ext
          else:
            - switch.turn_on: wifi_antenna_int_ext

sensor:
  - platform: ina226
    address: 0x41
    adc_averaging: 16
    max_current: 8.192A
    shunt_resistance: 0.005ohm
    bus_voltage:
      name: Battery Voltage
    current:
      name: Battery Current    

time:
  - platform: homeassistant
    id: homeassistant_time

touchscreen:
  - platform: gt911
    id: my_touchscreen
    interrupt_pin: GPIO23
    update_interval: never
    reset_pin:
      pi4ioe5v6408: pi4ioe1
      number: 5
    calibration:
      x_min: 0
      x_max: 720
      y_min: 0
      y_max: 1280

esp_ldo:
  - voltage: 2.5V
    channel: 3

color:
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%

font:
  - file: "gfonts://Roboto"
    id: roboto_32
    size: 32
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24
  - file: "gfonts://Roboto"
    id: roboto_12
    size: 12
  - file: "gfonts://Roboto"
    id: roboto_18
    size: 18

display:
  - platform: mipi_dsi
    rotation: 90
    id: my_display
    model: M5Stack-Tab5
    reset_pin:
      pi4ioe5v6408: pi4ioe1
      number: 4
#    pages:
#      - id: page1
#        lambda: |-
#          it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
#          it.rectangle(0, 20, it.get_width(), it.get_height(), id(my_blue));   // header bar
#          it.rectangle(0, 100, it.get_width(), it.get_height(), id(my_blue));  // footer bar          

output:
  - platform: ledc
    pin: GPIO22
    id: backlight_pwm
    frequency: 1000Hz

light:
  - platform: monochromatic
    output: backlight_pwm
    name: "Display Backlight"
    id: backlight
    restore_mode: RESTORE_DEFAULT_ON
    default_transition_length: 250ms

lvgl:
```

### Anything in the logs that might be useful for us?

```txt

```

RAR, 2025-09-22:
Alright, and colors were fixed with setting:

```
lvgl:
  byte_order: little_endian
```

clydebarrow, 2025-10-31:
Fixed by #11226