esphome/esphome #10760: ble_client disconnects prematurely (reason 0x13) when querying DIS characteristics

ngorchilov, 2025-09-16:
### The problem

**Description**

When using ESPHome’s ble_client on ESP32 (ESP-IDF) to query Device Information Service (DIS) characteristics (e.g., Manufacturer Name), the connection often closes prematurely with disconnect reason 0x13 (ESP_GATT_CONN_TERMINATE_PEER_USER).

Important: this is not an authorization issue.
- A correct value (e.g., "Grundfos Holding A/S") does come through from time to time.
- A simple Python BleakClient script is able to extract the same DIS characteristics from the device reliably every time.
- This points to a bug or timing problem in ESPHome’s ble_client, not the peripheral itself.

As a result:
- Entities usually get empty strings,
- Occasionally they update correctly,
- In the next second the reconnect loop clears the good value again.

This makes the behavior inconsistent and unreliable.

**Minimal YAML to reproduce**
```
esphome:
  name: grundfos-lc-231
  friendly_name: Grundfos LC-231

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
  
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_client: VERY_VERBOSE
    ble_client: VERY_VERBOSE
    esp32_ble_tracker: VERY_VERBOSE
    text_sensor: DEBUG
    logger: NONE
    api: NONE
    ota: NONE
    wifi: NONE
    mdns: NONE
    sensor: NONE
    binary_sensor: NONE
    switch: NONE
    number: NONE
    climate: NONE
    light: NONE
    fan: NONE
    cover: NONE
    component: WARN
    app: WARN

api:

ota:
  - platform: esphome
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:

ble_client:
  - mac_address: E4:52:1E:B2:01:6A
    id: lc231
  
text_sensor:
  - platform: ble_client
     name: Manufacturer
[... 568 lines cut ...]
[image]

### Which version of ESPHome has the issue?

2025.8.4

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

ngorchilov, 2025-09-24:
@bdraco thanks. I'm awaiting delivery of an ETH module. Once it arrives mid-Oct will cross-check and confirm if that solved the problem,

ngorchilov, 2025-10-19:
I got M5Stack PoESP32 and connected the client via Ethernet. Unfortunately, there's no change.

The bellow yaml is able to harvest all values over the course of few minutes (as low as 2 to as high as 40+), but the BLE keeps disconnecting and reconnecting:
```
substitutions:
  name: grundfos-lc-231
  friendly_name: Grundfos LC-231
  grundfos_mac_address: E4:52:1E:B2:01:6A

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp32:
  board: m5stack-core-esp32
  framework:
    type: esp-idf

logger:
  level: INFO

api:

ota:
  - platform: esphome

mdns:
  disabled: false

ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk:
    pin:
      number: GPIO0
      ignore_strapping_warning: true
    mode: CLK_EXT_IN
  phy_addr: 1
  power_pin:
    number: GPIO5
    ignore_strapping_warning: true

esp32_ble_tracker:

ble_client:
  - mac_address: ${grundfos_mac_address}
    id: ble_client_id

text_sensor:
  - platform: ble_client
    name: Manufacturer
    ble_client_id: ble_client_id
    service_uuid: 180A
    characteristic_uuid: 2A29
    entity_category: diagnostic
    filters:
      - lambda: |-
          if (x == "")
            return {};
          else 
            return x;

  - platform: ble_client
    name: Model Number
    ble_client_id: ble_client_id
    service_uuid: 180A
    characteristic_uuid: 2A24
    entity_category: diagnostic
    filters:
      - lambda: |-
          if (x == "")
            return {};
          else 
            return x;

  - platform: ble_client
    name: Hardware Revision
    ble_client_id: ble_client_id
    service_uuid: 180A
    characteristic_uuid: 2A27
    entity_category: diagnostic
    filters:
      - lambda: |-
          if (x == "")
            return {};
          else 
            return x;

  - platform: ble_client
    name: Firmware Revision
    ble_client_id: ble_client_id
    service_uuid: 180A
    characteristic_uuid: 2A26
    entity_category: diagnostic
    filters:
      - lambda: |-
          if (x == "")
            return {};
          else 
            return x;

  - platform: ble_client
    name: Software Revision
    ble_client_id: ble_client_id
    service_uuid: 180A
    characteristic_uuid: 2A28
    entity_category: diagnostic
    filters:
      - lambda: |-
          if (x == "")
            return {};
          else 
            return x;

  - platform: ethernet_info
    ip_address:
      name: IP Address
      id: mcu_ip_address
      entity_category: diagnostic
      icon: mdi:ip-network
    mac_address:
      name: MAC Address
      id: mcu_mac_address
      entity_category: diagnostic
      disabled_by_default: true
      icon: mdi:lan
```

bdraco, 2025-10-19:
Can you try dev or ESP IDF 5.5.x ?

Also try enabling the ESP-IDF framework logs under `esp32:`

ngorchilov, 2025-10-20:
I'll need some time for dev / 5.5.x as i have to restore the working state of my homebrew based setup first.

In the mean time 2 log excerpts with a bit of esp-idf logging. log_level: DEBUG as VERBOSE is practically unusable

**pure non-working state**
```
[00:20:49.065][VV][esp32_ble_tracker:294]: gap_scan_result - event 0
[00:20:49.065][VV][esp32_ble_tracker:384]: Parse Result:
[00:20:49.065][VV][esp32_ble_tracker:403]:   Address: E4:52:1E:B2:01:6A (PUBLIC)
[00:20:49.065][VV][esp32_ble_tracker:406]:   RSSI: -49
[00:20:49.065][VV][esp32_ble_tracker:407]:   Name: 'Созопол Ка\xd0'
[00:20:49.065][VV][esp32_ble_tracker:415]:   Ad Flag: 6
[00:20:49.065][VV][esp32_ble_tracker:434]:   Service data:
[00:20:49.065][VV][esp32_ble_tracker:435]:     UUID: 0xFE5D
[00:20:49.065][VV][esp32_ble_tracker:436]:     Data: 03.01.00.30.02.00.30.00.01.6A (10)
[00:20:49.065][VV][esp32_ble_tracker:439]:   Adv data: 02.01.06.0D.16.5D.FE.03.01.00.30.02.00.30.00.01.6A.15.09.D0.A1.D0.BE.D0.B7.D0.BE.D0.BF.D0.BE.D0.BB.20.D0.9A.D0.B0.D0.05.12.06.00.09.00 (45)
[00:20:49.065][D][esp32_ble_client:195]: [0] [E4:52:1E:B2:01:6A] Found device
[00:20:49.065][D][esp32_ble_tracker:639]: Found device E4:52:1E:B2:01:6A RSSI=-49
[00:20:49.065][D][esp32_ble_tracker:660]:   Address Type: PUBLIC
[00:20:49.065][D][esp32_ble_tracker:662]:   Name: 'Созопол Ка\xd0'
[00:20:49.065][V][esp32_ble_client:044]: [0] [E4:52:1E:B2:01:6A] Set state 3
[00:20:49.065][D][esp32_ble_tracker:131]: connecting: 0, discovered: 1, disconnecting: 0
[00:20:49.072][D][esp32_ble_tracker:779]: Stopping scan to make connection
[00:20:49.073][V][esp32_ble_client:044]: [0] [E4:52:1E:B2:01:6A] Set state 4
[00:20:49.080][V][esp32_ble:376]: gap_event_handler - 18
[00:20:49.083][V][esp32_ble_tracker:340]: gap_scan_stop_complete - status 0
[00:20:49.094][D][esp32_ble_tracker:741]: Scan stop complete, set scanner state to IDLE.
[00:20:49.095][D][esp32_ble_tracker:131]: connecting: 1, discovered: 0, disconnecting: 0
[00:20:49.236][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 40
[00:20:49.236][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=40 gattc_if=3
[00:20:49.238][D][esp32_ble_client:199]: [0] [E4:52:1E:B2:01:6A] ESP_GATTC_CONNECT_EVT
[00:20:49.241][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 2
[00:20:49.242][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=2 gattc_if=3
[00:20:49.249][D][esp32_ble_client:199]: [0] [E4:52:1E:B2:01:6A] ESP_GATTC_OPEN_EVT
[00:20:49.261][V][esp32_ble_client:044]: [0] [E4:52:1E:B2:01:6A] Set state 5
[00:20:49.261][I][esp32_ble_client:321]: [0] [E4:52:1E:B2:01:6A] Connection open
[00:20:49.269][D][esp32_ble_client:195]: [0] [E4:52:1E:B2:01:6A] Searching for services
[00:20:49.280][I][ble_text_sensor:041]: [Manufacturer] Connected successfully!
[00:20:49.280][I][ble_text_sensor:041]: [Model Number] Connected successfully!
[00:20:49.290][I][ble_text_sensor:041]: [Hardware Revision] Connected successfully!
[00:20:49.301][I][ble_text_sensor:041]: [Firmware Revision] Connected successfully!
[00:20:49.301][I][ble_text_sensor:041]: [Software Revision] Connected successfully!
[00:20:49.310][D][esp32_ble_tracker:131]: connecting: 0, discovered: 0, disconnecting: 0
[00:20:49.321][D][esp32_ble_tracker:207]: Starting scan, set scanner state to STARTING.
[00:20:49.331][V][esp32_ble:376]: gap_event_handler - 2
[00:20:49.331][V][esp32_ble_tracker:311]: gap_scan_set_param_complete - status 0
[00:20:49.334][V][esp32_ble:376]: gap_event_handler - 7
[00:20:49.341][V][esp32_ble_tracker:321]: gap_scan_start_complete - status 0
[00:20:49.716][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 46
[00:20:49.718][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=46 gattc_if=3
[00:20:49.719][D][esp32_ble_client:506]: [0] [E4:52:1E:B2:01:6A] Event 46
[00:20:49.739][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 7
[00:20:49.739][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=7 gattc_if=3
[00:20:49.744][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 7
[00:20:49.745][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=7 gattc_if=3
[00:20:49.749][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 7
[00:20:49.756][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=7 gattc_if=3
[00:20:49.764][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 7
[00:20:49.776][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=7 gattc_if=3
[00:20:49.776][V][esp32_ble:351]: gattc_event [esp_gatt_if: 3] - 6
[00:20:49.795][V][esp32_ble_client:267]: [0] [E4:52:1E:B2:01:6A] gattc_event_handler: event=6 gattc_if=3
[00:20:49.797][D][esp32_ble_client:199]: [0] [E4:52:1E:B2:01:6A] ESP_GATTC_SEARCH_CMPL_EVT
[00:20:49.797][V][esp32_ble_client:418]: [0] [E4:52:1E:B2:01:6A] Service UUID: 0x1800
[00:20:49.806][V][esp32_ble_client:420]: [0] [E4:52:1E:B2:01:6A]  start_handle: 0x1  end_handle: 0x7
[00:20:49.816][V][esp32_ble_client:418]: [0] [E4:52:1E:B2:01:6A] Service UUID: 0x1801
[00:20:49.816][V][esp32_ble_client:420]: [0] [E4:52:1E:B2:01:6A]  start_handle: 0x8  end_handle: 0x8
[00:20:49.825][V][esp32_ble_client:418]: [0] [E4:52:1E:B2:01:6A] Service UUID: 0x180A
[00:20:49.838][V][esp32_ble_client:420]: [0] [E4:52:1E:B2:01:6A]  start_handle: 0x9  end_handle: 0x13
[00:20:49.846][V][esp32_ble_client:418]: [0] [E4:52:1E:B2:01:6A] Service UUID: 0xFE5D
[... cut ...]
[00:21:28.729][VV][text_sensor.filter:040]: LambdaFilter(0x3ffbba04)::new_value() -> 
[00:21:28.739][VV][text_sensor.filter:013]: Filter(0x3ffbbaf8)::input()
[00:21:28.740][VV][text_sensor.filter:040]: LambdaFilter(0x3ffbbaf8)::new_value() -> 
[00:21:28.744][VV][esp-idf:000][BTU_TASK]: W (64024) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13 dev_find 1
[00:21:28.995][W][ble_text_sensor:121]: [Model Number] Cannot poll, not connected
[00:21:29.069][W][ble_text_sensor:121]: [Software Revision] Cannot poll, not connected
```

bdraco, 2025-10-20:
Can you try https://github.com/esphome/esphome/pull/11410 ?

ngorchilov, 2025-10-20:
Seems a lot better - it's able to read all characteristics at once. it's successful much more often - every second or so, but still alternates between successful reads and empty values.

here's a full log from running the client few minutes:
[logs_grundfos-lc-231_run.txt](https://github.com/user-attachments/files/23000575/logs_grundfos-lc-231_run.txt)

closed 2025-11-03 by jesserockz