esphome/esphome #10775: 2025.9.0: compile/upload ok but error in logging

Hermarcel, 2025-09-17:
### Version

2025.9.0

### What type of installation are you using?

Home Assistant Add-on

### Browser

Chrome

### What happened?

Compiling went fine.  Uploading thru the network, too. 
But then this appears:

```
Uploading: [============================================================] 100% Done...

INFO Upload took 9.15 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 10, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1269, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1256, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 716, in command_run
    return show_logs(config, args, devices)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 580, in show_logs
    if port_type in ("NETWORK", "MQTT") and has_mqtt_logging():
                                            ^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 215, in has_mqtt_logging
    return log_topic[CONF_LEVEL] != "NONE"
           ~~~~~~~~~^^^^^^^^^^^^
KeyError: 'level'
```

Worked fine in previous (2025.8."latest") version

### How to reproduce

Looks like it has something to do with mqtt logging (no api: entry in my yaml). 
When I enable native api: in my .yaml, the error disappears.
Removing tha api: entry makes the error re-appear.

There is a mqtt: entry in my .yaml:
```
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  log_topic: "${name}/log"
```

### Expected behavior

Do not generate this error and log to mqtt as before.

### Relevant log output

```shell
Uploading: [============================================================] 100% Done...

INFO Upload took 9.15 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 10, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1269, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1256, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 716, in command_run
    return show_logs(config, args, devices)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 580, in show_logs
    if port_type in ("NETWORK", "MQTT") and has_mqtt_logging():
                                            ^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 215, in has_mqtt_logging
    return log_topic[CONF_LEVEL] != "NONE"
           ~~~~~~~~~^^^^^^^^^^^^
KeyError: 'level'
```

ssieb, 2025-09-17:
As a workaround, you can add the level to the topic config.

ssieb, 2025-09-17:
Caused by https://github.com/esphome/esphome/pull/10632

closed 2025-09-17 by jesserockz