esphome/esphome #10819: I2C transaction unexpected nack detected errors on 2025.9.1

JeffResc, 2025-09-21:
### The problem

I have been experiencing I2C errors ever since upgrading from `2025.7.5`. I tried upgrading to `2025.9.1`, but I am getting various I2C errors ever since preventing me from using the device. I have not changed the YAML config since attempting the upgrade.

### Which version of ESPHome has the issue?

2025.9.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

api/i2c

### YAML Config

```yaml
esphome:
  name: emporia-vue-3
  friendly_name: emporia-vue-3

external_components:
#  - source: github://emporia-vue-local/esphome@dev
  - source: github://digiblur/esphome-vue3@dev
    components: [emporia_vue]

esp32:
  board: esp32dev
  framework:
    type: esp-idf

preferences:
  flash_write_interval: never

# Enable logging
logger:
  logs:
    sensor: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "[redacted]"

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: "192.168.2.12"
    gateway: "192.168.2.1"
    subnet: "255.255.255.0"
  ap:
    ssid: "Emporia-Vue-3 Fallback Hotspot"
    password: !secret fallback_password

captive_portal:

web_server:

prometheus:

i2c:
  sda: 5
  scl: 18
  scan: false
  frequency: 400kHz  # recommended range is 50-200kHz, was 200kHz but 150kHz stops errors (2024.8)
  timeout: 1ms
  id: i2c_a

switch:
  - platform: restart
    name: Restart

time:
  - platform: homeassistant
    id: my_time   

.defaultfilters:
  - &moving_avg
    # we capture a new sample every 0.24 seconds, so the time can
    # be calculated from the number of samples as n * 0.24.
    sliding_window_moving_average:
      # we average over the past 2.88 seconds
      window_size: 24
      # we push a new value every 1.44 seconds
      send_every: 12
  - &invert
    # invert and filter out any values below 0.
    lambda: 'return max(-x, 0.0f);'
  - &pos
    # filter out any values below 0.
    lambda: 'return max(x, 0.0f);'
  - &abs
    # take the absolute value of the value
    lambda: 'return abs(x);'

sensor:
  - platform: emporia_vue
    i2c_id: i2c_a
    phases:
      - id: phase_a
        input: BLACK
        calibration: 0.0194
        voltage:
          name: "Main Panel L1 Voltage"
          filters: [*moving_avg, *pos]
          state_class: measurement
          device_class: voltage
      - id: phase_b
        input: RED
        calibration: 0.0194
        voltage:
          name: "Main Panel L2 Voltage"
          filters: [*moving_avg, *pos]
          state_class: measurement
          device_class: voltage
    ct_clamps:
      # Main Panel Monitoring
      - phase_id: phase_a
        input: "A"
        power:
          name: "Main Panel L1 Power"
          id: phase_a_power
          device_class: power
          state_class: measurement
          filters: [*moving_avg, *pos]
      - phase_id: phase_b
        input: "B"
        power:
          name: "Main Panel L2 Power"
          id: phase_b_power
          device_class: power
          state_class: measurement
          filters: [*moving_avg, *pos]
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.9.1
INFO Reading configuration /config/emporia-vue-3.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Etc/UTC'
INFO Starting log output from 192.168.2.12 using esphome API
INFO Successfully resolved emporia-vue-3 @ 192.168.2.12 in 0.000s
INFO Successfully connected to emporia-vue-3 @ 192.168.2.12 in 0.026s
INFO Successful handshake with emporia-vue-3 @ 192.168.2.12 in 0.271s
[14:22:04.444][I][app:185]: ESPHome version 2025.9.1 compiled on Sep 21 2025, 12:57:31
[14:22:04.455][C][wifi:661]: WiFi:
[14:22:04.463][C][wifi:444]:   Local MAC: [redacted]
[14:22:04.468][C][wifi:449]:   SSID: [redacted]
[14:22:04.473][C][wifi:452]:   IP Address: 192.168.2.12
[14:22:04.477][C][wifi:456]:   BSSID: [redacted]
[14:22:04.477][C][wifi:456]:   Hostname: 'emporia-vue-3'
[14:22:04.477][C][wifi:456]:   Signal strength: -59 dB ââââ
[14:22:04.485][C][wifi:467]:   Channel: 11
[14:22:04.485][C][wifi:467]:   Subnet: 255.255.255.0
[14:22:04.485][C][wifi:467]:   Gateway: 192.168.2.1
[14:22:04.485][C][wifi:467]:   DNS1: 0.0.0.0
[14:22:04.485][C][wifi:467]:   DNS2: 0.0.0.0
[14:22:04.496][D][esp-idf:000]: E (412276) i2c.master: I2C hardware NACK detected
[14:22:04.504][D][esp-idf:000]: E (412285) i2c.master: I2C transaction unexpected nack detected
[14:22:04.522][D][esp-idf:000]: E (412292) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[14:22:04.530][D][esp-idf:000]: E (412299) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[14:22:04.539][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
[14:22:04.545][C][logger:273]: Logger:
[14:22:04.545][C][logger:273]:   Max Level: DEBUG
[14:22:04.545][C][logger:273]:   Initial Level: DEBUG
[14:22:04.548][C][logger:279]:   Log Baud Rate: 115200
[14:22:04.548][C][logger:279]:   Hardware UART: UART0
[14:22:04.552][C][logger:286]:   Task Log Buffer Size: 768
[14:22:04.556][C][logger:291]:   Level for 'sensor': INFO
[14:22:04.573][D][esp-idf:000]: E (412347) i2c.master: I2C hardware NACK detected
[14:22:04.573][D][esp-idf:000]: E (412355) i2c.master: I2C transaction unexpected nack detected
[14:22:04.573][D][esp-idf:000]: E (412358) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[14:22:04.577][D][esp-idf:000]: E (412363) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[14:22:04.586][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
[14:22:04.598][C][i2c.idf:081]: I2C Bus:
[14:22:04.608][C][i2c.idf:082]:   SDA Pin: GPIO5
[14:22:04.608][C][i2c.idf:082]:   SCL Pin: GPIO18
[14:22:04.608][C][i2c.idf:082]:   Frequency: 400000 Hz
[14:22:04.623][C][i2c.idf:088]:   Timeout: 1000us
[14:22:04.624][C][i2c.idf:092]:   Recovery: bus successfully recovered
[14:22:04.631][D][esp-idf:000]: E (412407) i2c.master: I2C hardware NACK detected
[14:22:04.635][D][esp-idf:000]: E (412420) i2c.master: I2C transaction unexpected nack detected
[14:22:04.640][D][esp-idf:000]: E (412425) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[14:22:04.648][D][esp-idf:000]: E (412429) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[14:22:04.661][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
[14:22:04.666][D][esp-idf:000]: E (412452) i2c.master: I2C hardware NACK detected
[14:22:04.670][D][esp-idf:000]: E (412457) i2c.master: I2C transaction unexpected nack detected
[14:22:04.681][D][esp-idf:000]: E (412460) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[14:22:04.692][D][esp-idf:000]: E (412468) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[14:22:04.701][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
```

### Additional information

I looked at #10346 who reported other I2C errors, but the fix PR seems to already be merged in `2025.9.1`. I also followed [this](https://discord.com/channels/429907082951524364/1407853598821449928) Discord thread in attempting to change the esp-idk version to `5.3.2`, but I got an error that the I2C component needs at least version `5.4.2`.

swoboda1337, 2025-09-22:
You can't use an older IDF version support was removed for that in the I2C code. 

Does a lower speed fix it? Not sure what your comments are saying

JeffResc, 2025-09-22:
Hey @swoboda1337, thanks for the quick reply!

I think you may have removed the original comment so this may be irrelevant, but I tried it with the emporia_vue branch of your fork, which compiled and flashed, but I got a wall of `Failed to read from sensor due to a malformed reading, should end in null bytes but is 65535` at runtime at 400kHz.

Also, sorry about the comment in the code, I get that can be confusing. I didn't write that comment, rather it came from the upstream template I based my config on. I have always used 400kHz prior to this error and haven't had any problems in past versions. Additionally, I just now tried it with 50kHz and I am getting similar results:
```
[03:51:05.045][D][esp-idf:000]: E (5195) i2c.master: I2C hardware NACK detected
[03:51:05.055][D][esp-idf:000]: E (5204) i2c.master: I2C transaction unexpected nack detected
[03:51:05.079][D][esp-idf:000]: E (5214) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[03:51:05.223][D][esp-idf:000]: E (5224) i2c.master: i2c_master_execute_defined_operations(1366): I2C transaction failed
[03:51:06.366][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
```

JeffResc, 2025-09-22:
I just tried that and I'm getting
```
INFO ESPHome 2025.9.1
INFO Reading configuration /config/emporia-vue-3.yaml...
Failed config

esp32: [source /config/emporia-vue-3.yaml:11]
  board: esp32dev
  
  ESP-IDF 5.5.1 is not available with pioarduino; you may need to specify 'release'.
  framework: 
    type: esp-idf
    version: 5.5.1
    platform_version: 55.03.31
```

I'm currently not sure why it's saying `ESP-IDF 5.5.1 is not available with pioarduino` because to my understanding, ESP-IDF and pioarduino are mutually exclusive frameworks. I'll keep digging into this, but figured I would let you know in the meantime while I investigated

Saviq, 2025-09-27:
I was setting up a completely new Vue 3 and this is what finally made it work for me:

https://dracoli.ch/posts/vue3-esphome-ha/

FlyAndy123, 2025-10-02:
@swoboda1337 I realized yesterday that my Emporia Vue 3 was also not working in 2025.9. I've tried all of the above including reducing the update_interval to 1s. This still has no effect, it is not reading anymore.
I've managed to roll back to 2025.7.5 which seems to be the last version before there were issues.

jtweaver, 2025-11-28:
Has anyone found a solution for this (without downgrading)? I'm having the same issue on 2025.11.2 and none of the suggestions above seem to work.

swoboda1337, 2025-11-29:
Just to confirm increasing the update_interval was tried:
```
sensor:
  - platform: emporia_vue
    i2c_id: i2c_a
    update_interval: 2s
```

jtweaver, 2025-11-29:
> Just to confirm increasing the update_interval was tried:
> 
> ```
> sensor:
>   - platform: emporia_vue
>     i2c_id: i2c_a
>     update_interval: 2s
> ```

I did. The only thing that changes is the frequency of the failure logs.

```
[22:36:19.840][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
[22:36:21.833][D][esp-idf:000]: E (92695) i2c.master: I2C hardware NACK detected
[22:36:21.835][D][esp-idf:000]: E (92697) i2c.master: I2C transaction unexpected nack detected
[22:36:21.877][D][esp-idf:000]: E (92700) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
[22:36:21.883][D][esp-idf:000]: E (92703) i2c.master: i2c_master_execute_defined_operations(1401): I2C transaction failed
[22:36:21.883][E][emporia_vue:048]: Failed to read from sensor due to I2C error 2
```

jtweaver, 2025-11-30:
> No valid data is ever read? Did you try to lower the I2C frequency to like 100khz or less (the pull ups seem really weak)?
> 
> EDIT: could also try to enable scan

Yeah I tried 100khz with and without scan: true. Also tried 150khz up to 400khz.

```
i2c:
  sda: 5
  scl: 18
  scan: true
  # recommended range is 50-200kHz
  frequency: 100kHz
  timeout: 1ms
  id: i2c_a
```

swoboda1337, 2026-01-12:
Prefect, 2026.1 (released next week) will move to that platform.

devmaximilian, 2026-01-15:
The breakout modules I'm using have onboard pullup resistors of 4.7kΩ for sda. It could be that they were causing the issue I was facing given the warning displayed for both pin 8 and 9.

"WARNING GPIO8 is a Strapping PIN and should be avoided. Attaching external pullup/down resistors to strapping pins can cause unexpected failures"