esphome/esphome #10820: Waveshare_epaper compilation problem on esphome 2025.9.x

WarC0zes, 2025-09-21:
### The problem

Hi,
I wanted to update a reTerminal E1002 and I get this error during compilation. I can't compile the firmware. In version 2025.8.3, I have no problem.

### Which version of ESPHome has the issue?

2025.9.x

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

Waveshare_epaper

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs\reterminal-e1002\src\esphome\components\waveshare_epaper\waveshare_213v3.cpp.o
Compiling .pioenvs\reterminal-e1002\src\esphome\components\waveshare_epaper\waveshare_epaper.cpp.o
Compiling .pioenvs\reterminal-e1002\src\esphome\components\wifi\wifi_component.cpp.o
src/esphome/components/waveshare_epaper/waveshare_213v3.cpp: In member function 'virtual void esphome::waveshare_epaper::WaveshareEPaper2P13InV3::dump_config()':
src/esphome/components/waveshare_epaper/waveshare_213v3.cpp:185:1: error: expected ';' before '}' token
  185 | }
      | ^
Compiling .pioenvs\reterminal-e1002\src\esphome\components\wifi\wifi_component_esp_idf.cpp.o
*** [.pioenvs\reterminal-e1002\src\esphome\components\waveshare_epaper\waveshare_213v3.cpp.o] Error 1
============================================= [FAILED] Took 129.20 seconds =============================================
```

### Additional information

It seems to come from this commit
https://github.com/esphome/esphome/pull/10636/commits/0f2a8300b290b5513336d26068e7964fa35de012#diff-967f4727cde2afa27e81f14211edbbec1219f23efee46b0e5e2365dcf43e897c

bdraco, 2025-09-21:
esphome/components/waveshare_epaper/waveshare_213v3.cpp must be from the external component. It seems they are missing a `;` in the code

bdraco, 2025-09-21:
Looks like someone already has a PR open to fix it https://github.com/lublak/esphome/pull/1

closed 2025-09-21 by WarC0zes