esphome/esphome #10877: ibrary Manager don`t downloading

strk90, 2025-09-25:
### The problem

When compiling the file, the "Library Manager" files aren't downloaded.
Please help.

### Which version of ESPHome has the issue?

ESPHome 2025.9.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### YAML Config

```yaml
esphome:
  name: "kalitka"
  friendly_name: "kalitka"

esp8266:
  board: d1_mini
  framework:
    version: recommended

# Enable logging
logger:

api:
  encryption:
    key: "zsBwRY2zv8EhlFtmdKw+1TFLnjFjOhnv7obuck7KZs4="
  
ota:
- platform: esphome
  password: "142b32ced0f60a2ba4d8faf0257acde0"
wifi:
  ssid: Keenetic-7693
  password: hMDhDAtx
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Scan Fallback Hotspot"
    password: "pTxA44xVXfCt"
captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2025.9.1
INFO Reading configuration /config/esphome/kalitka.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing kalitka (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
Library Manager: Installing ESP32Async/ESPAsyncTCP @ 2.0.0
INFO Installing ESP32Async/ESPAsyncTCP @ 2.0.0
Downloading  [------------------------------------]    0%
```

Roman-ENGINEER, 2025-10-22:
Точно такая же проблема. Смог кто-нибудь решить? Подскажите пожалуйста. Удалял уже все - не помогает.

akintash, 2025-10-22:
the same happened today

NeurName, 2025-10-24:
> I'm having a similar problem. Various add-ons like Samba Share won't launch, and when installing a new ESP32 or ESP8266 controller, some libraries won't download. It's a bit frustrating.=(

The simple answer is Roskomnadzor. Russia is being cut off from the internet, creating a local network with no access to the outside world. Finding a VPN or moving to another country might be a temporary solution. I also couldn't find a way to download and install libraries and dependencies without using ESPHome.

desolaris, 2026-01-02:
I got around the problem by running traffic from the domain platformio.org through a VPN located in the EU zone

yar-zaytsev, 2026-01-06:
> Same problem:
> 
> ```
> INFO ESPHome 2025.9.2
> INFO Reading configuration /config/bme280_1.yaml...
> INFO Generating C++ source...
> INFO Compiling app...
> Processing bme280-1 (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
> --------------------------------------------------------------------------------
> Library Manager: Installing ESP32Async/ESPAsyncTCP @ 2.0.0
> INFO Installing ESP32Async/ESPAsyncTCP @ 2.0.0
> Downloading  [------------------------------------]    0%
> ```

resolve:
1. get name docker container name of addon:
```
>#docker ps | grep esphome
```

[image]

2. Download from github libraries :
https://github.com/ESP32Async/ESPAsyncWebServer/archive/refs/tags/v3.7.10.zip
3. Copy unarchived folder to server via scp 

[image]

4. create file .piopm in each library with this example data. For ESPAsyncTCP:

{"type": "library", "name": "ESPAsyncTCP", "version": "2.0.0", "spec": {"owner": "esp32async", "id": 18272, "name": "ESPAsyncTCP", "requirements": null, "uri": null}}

[image]

5. Copy folder to yours project dir in docker container /data/build/project_name/.piolibdeps......
```
docker cp ./ESPAsyncTCP addon_5c53de3b_esphome:/data/build/esp8266/.piolibdeps/esp8266/ESPAsyncTCP
```

[image]
6. Repeat for other libs.

7. Compile

[image]

closed 2026-04-14 by github-actions[bot]