esphome/esphome #10880: ESP32-C3 + ili9xxx (ILI9341 over SPI) shows black screen without errors; works fine with Arduino and ESP32-S3

14790897, 2025-09-25:
### The problem

Screen stays white, with no output at all.

Logs show no errors (display component does not report any failure).
```
INFO Starting log output from COM38 with baud rate 115200
```

### Which version of ESPHome has the issue?

Version: 2025.9.1

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

ili9xxx

### YAML Config

```yaml
esphome:
  name: c3-ili9341-test

esp32:
  board: airm2m_core_esp32c3
  framework:
    type: arduino

logger:
  level: DEBUG

wifi:
  ssid: "test"
  password: "test12345678"

api:

spi:
  id: bus_a
  clk_pin: GPIO2
  mosi_pin: GPIO3
  # miso_pin not connected

display:
  - platform: ili9xxx
    spi_id: bus_a
    model: TFT 2.4
    cs_pin: GPIO7
    dc_pin: GPIO6
    reset_pin: GPIO10
    data_rate: 10MHz
    rotation: 0
    invert_colors: false
    lambda: |-
      it.fill(Color(0, 0, 64));               // Deep blue background
      it.filled_rectangle(10, 10, 100, 50, Color(255, 255, 255));  // White
      it.filled_rectangle(20, 70, 60, 30, Color(255, 0, 0));       // Red
      it.filled_rectangle(90, 120, 80, 40, Color(0, 255, 0));      // Green
```

### Anything in the logs that might be useful for us?

```txt

```

### Additional information

Arduino Sketch (Working for Comparison)
This sketch works with the exact same wiring (Adafruit_ILI9341 + hardware SPI):
```
#include <Arduino.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ILI9341.h>
#include <SPI.h>

const uint8_t TFT_CS = 7;    // CS
const uint8_t TFT_DC = 6;    // DC
const uint8_t TFT_RST = 10;  // RST

Adafruit_ILI9341 tft(TFT_CS, TFT_DC, TFT_RST);

void setup() {
  Serial.begin(115200);
  tft.begin();
  tft.setRotation(1);
  tft.fillScreen(ILI9341_BLACK);
  tft.setTextColor(ILI9341_WHITE); tft.setTextSize(2);
  tft.setCursor(10, 10); tft.print("ILI9341 Test");
  tft.setCursor(10, 30); tft.print("ESP32-C3");
  tft.setCursor(10, 50); tft.setTextColor(ILI9341_GREEN);
  tft.print("320x240 TFT");
}

void loop() {}

```

14790897, 2025-09-28:
@esphomebot

swoboda1337, 2025-09-30:
Did you try setting the model to `model: ILI9341` 

See: https://esphome.io/components/display/ili9xxx/

swoboda1337, 2025-09-30:
Aren't your dimensions swapped should be 240x320