esphome/esphome #10896: One Wire on esp32c6

Courty40, 2025-09-26:
### The problem

Adding OneWire to a esp32c6 give errors :

INFO ESPHome 2025.9.1
INFO Reading configuration /config/esphome/esphome-web-7c6f30.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 10, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1269, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1256, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 677, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 364, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 371, in generate_cpp_contents
    for name, component, conf in iter_component_configs(CORE.config):
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 61, in iter_component_configs
    p_name = f"{domain}.{p_config[CONF_PLATFORM]}"
                         ~~~~~~~~^^^^^^^^^^^^^^^
TypeError: 'int' object is not subscriptable

### Which version of ESPHome has the issue?

2025.9.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

One Wire

### YAML Config

```yaml
esphome:
  name: esphome-web-7c6f30
  friendly_name: ESPHome Web 7c6f30
  min_version: 2025.9.0
  name_add_mac_suffix: false

esp32:
  variant: esp32c6
  framework:
    type: esp-idf

logger:

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

one_wire:
  - platform: gpio
    pin: GPIO10
```

### Anything in the logs that might be useful for us?

```txt
No Logs as wont compile
```

swoboda1337, 2025-09-26:
Its not a one_wire issue, you are missing the platform for ota:
```
ota:
  - platform: "esphome"
```
There is an issue because it should print a meaningful error and not fail like that.

swoboda1337, 2025-09-26:
@bdraco I think this is because you added `ota.web_server` to `AUTO_LOAD` in `captive portal`. 

This check fails in `ota/__init.py`:
```
def _ota_final_validate(config):
    if len(config) < 1:
        raise cv.Invalid(
            f"At least one platform must be specified for '{CONF_OTA}'; add '{CONF_PLATFORM}: {CONF_ESPHOME}' for original OTA functionality"
        )
```

Because the config looks like this when captive portal is included:
```
{0: {'platform': 'web_server', 'id': ID<web_server_webserverotacomponent_id declaration=True, type=web_server::WebServerOTAComponent, manual=False>}}
```
and its not a list like it should be:
```
[{'platform': 'esphome', 'id': ID<esphome_esphomeotacomponent_id declaration=True, type=esphome::ESPHomeOTAComponent, manual=False>, 'version': 2, 'port': 3232}]
```

bdraco, 2025-09-26:
```
DEBUG: LoadValidationStep.run - esphome: conf={'name': 'esphome-web-7c6f30', 'friendly_name': 'ESPHome Web 7c6f30', 'min_version': '2025.9.0', 'na, type=<class 'esphome.helpers.OrderedDict'>
DEBUG: LoadValidationStep.run - esphome: stored in result as {'name': 'esphome-web-7c6f30', 'friendly_name': 'ESPHome Web 7c6f30', 'min_version': '2025.9.0', 'na
DEBUG: LoadValidationStep.run - esp32: conf={'variant': 'esp32c6', 'framework': {'type': 'esp-idf'}}, type=<class 'esphome.helpers.OrderedDict'>
DEBUG: LoadValidationStep.run - esp32: stored in result as {'variant': 'esp32c6', 'framework': {'type': 'esp-idf'}}
DEBUG: LoadValidationStep.run - logger: conf=None, type=<class 'NoneType'>
DEBUG: LoadValidationStep.run - logger: stored in result as None
DEBUG: LoadValidationStep.run - api: conf=None, type=<class 'NoneType'>
DEBUG: LoadValidationStep.run - api: stored in result as None
DEBUG: LoadValidationStep.run - ota: conf=None, type=<class 'NoneType'>
DEBUG: LoadValidationStep.run - ota: stored in result as None
DEBUG: LoadValidationStep - ota is platform component, normalizing conf: None
DEBUG: LoadValidationStep - ota was falsy, converted to []
DEBUG: LoadValidationStep.run - wifi: conf={'ssid': 'rocketiot', 'password': 'paneswapnicelyhandbag'}, type=<class 'esphome.helpers.OrderedDict'>
DEBUG: LoadValidationStep.run - wifi: stored in result as {'ssid': 'rocketiot', 'password': 'paneswapnicelyhandbag'}
DEBUG: LoadValidationStep.run - captive_portal: conf=None, type=<class 'NoneType'>
DEBUG: LoadValidationStep.run - captive_portal: stored in result as None
DEBUG: LoadValidationStep.run - one_wire: conf=[{'platform': 'gpio', 'pin': 'GPIO10'}], type=<class 'esphome.helpers.EList'>
DEBUG: LoadValidationStep.run - one_wire: stored in result as [{'platform': 'gpio', 'pin': 'GPIO10'}]
DEBUG: LoadValidationStep - one_wire is platform component, normalizing conf: [{'platform': 'gpio', 'pin': 'GPIO10'}]
DEBUG: Before assignment - one_wire: type=<class 'esphome.helpers.EList'>, value=[{'platform': 'gpio', 'pin': 'GPIO10'}]
DEBUG: After assignment [0] - one_wire: type=<class 'esphome.helpers.EList'>, value=[{'platform': 'gpio', 'pin': 'GPIO10'}]
DEBUG: _process_platform_config - one_wire before: type=<class 'esphome.helpers.EList'>, value=[{'platform': 'gpio', 'pin': 'GPIO10'}]
DEBUG: LoadValidationStep.run - preferences: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - preferences: stored in result as {}
DEBUG: LoadValidationStep.run - socket: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - socket: stored in result as {}
DEBUG: LoadValidationStep.run - md5: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - md5: stored in result as {}
DEBUG: LoadValidationStep.run - safe_mode: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - safe_mode: stored in result as {}
DEBUG: LoadValidationStep.run - network: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - network: stored in result as {}
DEBUG: LoadValidationStep.run - web_server_base: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - web_server_base: stored in result as {}
DEBUG: AutoLoad ota.web_server - component ota config before: [], type: <class 'list'>
DEBUG: Before append - ota: component_conf type=<class 'list'>, value=[]
DEBUG: Before append - result[ota]: type=<class 'list'>, value=[]
DEBUG: After append - ota: component_conf type=<class 'list'>, value=[{'platform': 'web_server'}]
DEBUG: After append - result[ota]: type=<class 'list'>, value=[{'platform': 'web_server'}]
DEBUG: _process_platform_config - ota before: type=<class 'list'>, value=[{'platform': 'web_server'}]
DEBUG: LoadValidationStep.run - mdns: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - mdns: stored in result as {}
DEBUG: LoadValidationStep.run - web_server_idf: conf={}, type=<class 'esphome.core.AutoLoad'>
DEBUG: LoadValidationStep.run - web_server_idf: stored in result as {}

```

Something is converting it back to a dict after its converted to a list, but I can't find it

bdraco, 2025-09-26:
normalization happens too late

Courty40, 2025-09-27:
> > It shouldn't error at all, its a valid config to load `ota` with no platforms. Not a good idea but valid
> 
> Well it's crashing instead of throwing an error atm :)
> 

Thanks guys.. this is literally how the docs say to find the id of the one wire device before you enter the full config (id appears in the logs)

closed 2025-09-28 by jesserockz