esphome/esphome #10923: Radar LD2410C issues after updating to 2025.9

TioBundy, 2025-09-28:
### The problem

I've been experiencing varying performance with this radar with Esphome versions, but this isn't normal anymore. It's always performed more or less responsively and immediately, but correctly.

Since Esphome 2025.9, false readings (turning on when no one is there) or, after turning on, maintaining high "Still Target" or "Moving Target" readings, often exceeding 60-70%, cause the automation controlling an LED strip to never turn off, remaining on for up to 9 hours, like tonight.

Restarting the device works, keeping the values ​​at 0% or almost 0%, but it starts happening again soon after. Many times, returning to the room to stay or move around, it updates the values ​​and works again, but it's still problematic because the operation is completely contrary to what you're trying to achieve, even becoming more annoying than a light that you press when leaving the room.

Please, I should be able to fix this, it's not normal for these changes to suddenly happen without anything like this happening before.

### Which version of ESPHome has the issue?

2025.9

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

ld2410

### YAML Config

```yaml
substitutions:
  name: presencia-cocina-tira-led
  friendly_name: Presencia Cocina (Tira Led)

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev
  framework:
    type: esp-idf

#esp32_ble_tracker:
  #scan_parameters:
    #interval: 1100ms
    #window: 500ms
    #active: true
#bluetooth_proxy:
  #active: true
  #connection_slots: 5

logger: 

api:
  encryption:
    key: b0JSh74MYXNHCdok6qajnRPUHDbiiok0b2i01jIu8b8=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.203
    gateway: 192.168.1.1
    subnet: 255.255.255.0

ota:
- platform: esphome   
#SERVIDOR WEB ESPHOME
#web_server:
  #port: 80
  #version: 3

# INCLUSION DEL SENSOR PRESENCIA LD2410c
uart:
  tx_pin: GPIO16
  rx_pin: GPIO17
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  id: SENSORCOCINA
# Sensor presencia mmWave
sensor:
  - platform: ld2410
    moving_distance:
      name : Moving Distance
      id: my_movingdistance
    still_distance:
      name: Still Distance
      id: my_stilldistance
    moving_energy:
      name: Move Energy
      id: my_movingenergy
    still_energy:
      name: Still Energy
      id: my_stillenergy
    detection_distance:
      name: Detection Distance
      id: my_detectiondistance
    g0:
      move_energy:
        name: g0 move energy
      still_energy:
        name: g0 still energy
    g1:
      move_energy:
        name: g1 move energy
      still_energy:
        name: g1 still energy
    g2:
      move_energy:
        name: g2 move energy
      still_energy:
        name: g2 still energy
    g3:
      move_energy:
        name: g3 move energy
      still_energy:
        name: g3 still energy
    g4:
      move_energy:
        name: g4 move energy
      still_energy:
        name: g4 still energy
    g5:
      move_energy:
        name: g5 move energy
      still_energy:
        name: g5 still energy
    g6:
      move_energy:
        name: g6 move energy
      still_energy:
        name: g6 still energy
    g7:
      move_energy:
        name: g7 move energy
      still_energy:
        name: g7 still energy
    g8:
      move_energy:
        name: g8 move energy
      still_energy:
        name: g8 still energy
binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
      id: my_hastarget
    has_moving_target:
      name: Moving Target
      id: my_movingtarget
    has_still_target:
      name: Still Target
      id: my_hasstilltarget
    out_pin_presence_status:
      name: out pin presence status
      id: my_presencestatus

number:
  - platform: ld2410
    timeout:
      name: timeout
    #light_threshold:
      #name: light threshold
    max_move_distance_gate:
      name: Distancia Máxima Movimiento
    max_still_distance_gate:
      name: Distancia Máxima Presencia
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold
        
# Botón Reinicio, etc
button:
  - platform: ld2410
    #restart: 
      #name: 'Reiniciar Led Cocina'
    factory_reset:
      name: 'Factory reset'

# Botón Reinicio
  - platform: restart
    name: 'Reiniciar Presencia Cocina'
```

### Anything in the logs that might be useful for us?

```txt

```

danielholm, 2025-10-01:
Try with esp-idf version 5.4.1. See here: https://github.com/esphome/esphome/issues/10608

TioBundy, 2025-10-05:
Hello guys, try this: 

Simply install the esphome legacy add-on plugin repo:

https://github.com/khenderick/esphome-legacy-addons

In my case I have compiled the conflicting board with "2025.7.5" and everything works without flaws, but since I lost my settings of how and when I detect my LD2410c it is a little less effective now, but no It fails definitely. 

What I do not know is how that plugin detects the plates that are already online, the one that gave problems detects it well, the others have different names as recently installed, but I did not investigate Plus, maybe they're the ones I have running with the generic name of esphome, a matter of looking at the Yaml and the IP they use.

jcd420, 2025-10-15:
> Hi,
> 
> I can confirm that — it works for me as well with the following code using ESPHome 2025.9.3:
> 
> ```
> esp32:
>   board: esp32-c3-devkitm-1
>   variant: ESP32C3
>   framework:
>     type: esp-idf
>     version: 5.2.2
>     platform_version: 6.7.0
> ```

I get the follow erro:

Failed config

i2c: [source /config/esphome/espsuit2bath.yaml:18]
  
  This feature requires at least framework version 5.4.2.
  - id: bus_a
    sda: 21
    scl: 22
    timeout: 5s
  - id: bus_b
    sda: 16
    scl: 17
    scan: True

TioBundy, 2025-11-07:
> Hi. I tried with 2025.10.4 the problem seems to be solved. I suggest you to check it

Nothing is lost to try. I had to reflash the board with the version from a few months ago, but it was working perfectly so far.

closed 2026-02-13 by github-actions[bot]