esphome/esphome #10925: [usb_host] Crash due to over-free

clydebarrow, 2025-09-28:
### The problem

Device using usb_uart and modbus now crashes every few minutes since #10859.

Crash log below.

### Which version of ESPHome has the issue?

2025.10-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

usb_uart

### YAML Config

```yaml
substitutions:
  friendly_name: Power Meter
  name: power-meter

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  platformio_options:
    build_unflags: -Werror=all
    board_build.flash_mode: dio

esp32:
  board: esp32-s3-devkitc-1
  flash_size: 8MB
  cpu_frequency: 240MHz
  framework:
    type: esp-idf

api:
  encryption:
    key: !secret encryption_key

logger:
  level: DEBUG
  hardware_uart: UART0
  logs:
    wifi: INFO
    modbus_controller.sensor: INFO
    modbus_controller: INFO
    sensor: INFO

wifi: !include wifi.yaml

sensor:
  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db
    update_interval: 10s
    entity_category: "diagnostic"
  - platform: debug
    free:
      name: "Heap Free"
    loop_time:
      name: "Loop Time"
    psram:
[... 163 lines cut ...]
WARNING Decoded 0x420cb915: std::_Function_handler<void (), esphome::Component::status_momentary_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long)::{lambda()#1}>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/clyde/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:285
WARNING Decoded 0x4200edb2: esphome::modbus::Modbus::parse_modbus_byte_(unsigned char) at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/modbus/modbus.cpp:147
WARNING Decoded 0x4201069c: esphome::modbus_controller::ModbusController::on_modbus_write_registers(unsigned char, std::vector<unsigned char, std::allocator<unsigned char> > const&) at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/modbus_controller/modbus_controller.cpp:223
WARNING Decoded 0x420108d7: esphome::network::is_connected() at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/network/util.cpp:50
WARNING Decoded 0x4200f0f1: esphome::modbus::ModbusDevice::send(unsigned char, unsigned short, unsigned short, unsigned char, unsigned char const*) at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/modbus/modbus.h:65
 (inlined by) esphome::modbus_controller::ModbusCommandItem::send() at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/modbus_controller/modbus_controller.cpp:645
WARNING Decoded 0x4200f165: esphome::modbus_controller::ModbusCommandItem::is_equal(esphome::modbus_controller::ModbusCommandItem const&) at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/modbus_controller/modbus_controller.cpp:662 (discriminator 4)
WARNING Decoded 0x420cc1c9: std::_Function_handler<unsigned short (unsigned short), setup()::{lambda(unsigned short)#23}>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/clyde/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/std_function.h:270
WARNING Decoded 0x420176dd: void esphome::ComponentIterator::process_platform_item_<esphome::StaticVector<esphome::button::Button*, 2u> >(esphome::StaticVector<esphome::button::Button*, 2u> const&, bool (esphome::ComponentIterator::*)(esphome::StaticVector<esphome::button::Button*, 2u>::value_type)) at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/core/component_iterator.h:182 (discriminator 1)
 (inlined by) esphome::ComponentIterator::advance() at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/core/component_iterator.cpp:74 (discriminator 1)
WARNING Decoded 0x42016e2e: std::_Vector_base<std::pair<esphome::Component const*, char const*>, std::allocator<std::pair<esphome::Component const*, char const*> > >::_M_deallocate(std::pair<esphome::Component const*, char const*>*, unsigned int) at /Users/clyde/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:388
 (inlined by) std::_Vector_base<std::pair<esphome::Component const*, char const*>, std::allocator<std::pair<esphome::Component const*, char const*> > >::~_Vector_base() at /Users/clyde/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:368
 (inlined by) std::vector<std::pair<esphome::Component const*, char const*>, std::allocator<std::pair<esphome::Component const*, char const*> > >::~vector() at /Users/clyde/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/stl_vector.h:738
 (inlined by) std::default_delete<std::vector<std::pair<esphome::Component const*, char const*>, std::allocator<std::pair<esphome::Component const*, char const*> > > >::operator()(std::vector<std::pair<esphome::Component const*, char const*>, std::allocator<std::pair<esphome::Component const*, char const*> > >*) const at /Users/clyde/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/14.2.0/bits/unique_ptr.h:93
WARNING Decoded 0x4201a096: esphome::LambdaCondition<float>::LambdaCondition(std::function<bool (float)>&&) at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/core/base_automation.h:74
WARNING Decoded 0x420089ee: esphome::has_custom_mac_address() at /Users/clyde/dev/opensourceprojects/esp/my-devices/power-meter/.esphome/build/power-meter/src/esphome/components/esp32/helpers.cpp:98 (discriminator 1)
ERROR Serial port closed!
(venv) [macair-9:esp/my-devices/power-meter] clyde%
```

### Additional information

Appears to have been introduced with #10859.

Reverting `esphome/components/usb_{uart,host}` to e020110579d9ab60b7c2b1e5f8814cd32a48fb0e makes the issue go away.

clydebarrow, 2025-09-28:
Looks like the crash log has been decoded incorrectly due to using the wrong .elf file (serial debug being done on different machine to the one that was used to OTA flash.)

clydebarrow, 2025-09-28:
Attaching updated crash log

[xx.log](https://github.com/user-attachments/files/22585699/xx.log)

bdraco, 2025-09-28:
trq pool is accessed by different tasks. It needs a lock to protect it

bdraco, 2025-09-28:
Could protect it with a mutex. But we have lock free atomics and we know we already have a performance issue so atomics, while a bit more complex is probably the right solution here

closed 2025-09-29 by clydebarrow