esphome/esphome #10934: ESPHome cannot dowland any additional packages needed to build the firmware

kolodziejczyk123, 2025-09-29:
### The problem

INFO ESPHome 2025.9.1
INFO Reading configuration /config/esphome/falownik.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing deye-inverter (board: esp32dev; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Installing tools via idf_tools.py (this may take several minutes)...
ERROR idf_tools.py installation failed
INFO Package configuration completed successfully
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.2.1/esp32-3.2.1.zip
INFO Installing https://github.com/espressif/arduino-esp32/releases/download/3.2.1/esp32-3.2.1.zip
Tool Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
SSLError: Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 464, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1093, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 790, in connect
    sock_and_verified = _ssl_wrap_socket_and_match_hostname(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 969, in _ssl_wrap_socket_and_match_hostname
    ssl_sock = ssl_wrap_socket(
               ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/ssl_.py", line 480, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls, server_hostname)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/ssl_.py", line 524, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ssl.py", line 455, in wrap_socket
    return self.sslsocket_class._create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ssl.py", line 1041, in _create
    self.do_handshake()
  File "/usr/local/lib/python3.12/ssl.py", line 1319, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'release-assets.githubusercontent.com'. (_ssl.c:1010)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 488, in _make_request
    raise new_e
urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'release-assets.githubusercontent.com'. (_ssl.c:1010)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 644, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 841, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='release-assets.githubusercontent.com', port=443): Max retries exceeded with url: /github-production-release-asset/70127218/dcc097c8-0a75-4310-bb5b-a046ced1eae7?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-09-29T13%3A08%3A47Z&rscd=attachment%3B+filename%3Desp32-3.2.1.zip&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-09-29T12%3A07%3A54Z&ske=2025-09-29T13%3A08%3A47Z&sks=b&skv=2018-11-09&sig=hMTxm5UZe6z7dOXy01R0Yp8I34vQGezYx%2FPue0iVTVY%3D&jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1OTE0ODQ0NSwibmJmIjoxNzU5MTQ4MTQ1LCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.Kro7Pw5JmzWkp6Q3YOFn3uq7jvVRAgRks2bfy97ulmI&response-content-disposition=attachment%3B%20filename%3Desp32-3.2.1.zip&response-content-type=application%2Foctet-stream (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'release-assets.githubusercontent.com'. (_ssl.c:1010)")))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
[... 93 lines cut ...]
============================================================

### Which version of ESPHome has the issue?

2025.9.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### YAML Config

```yaml

```

### Anything in the logs that might be useful for us?

```txt

```

swoboda1337, 2025-09-30:
Did this resolve itself? Seems like a problem with your connection?

swoboda1337, 2025-10-01:
I am not sure why it fails, the only thing I can think of is to clean platformio.

If you are using the addon maybe try to uninstall the addon, click remove data and reinstall? This wont remove the yamls. 

Can try to remove ~/.platformio if you are using ESPHome locally.

SteinarSun, 2025-10-02:
I've got the same problem with cmake . 
Started after the updtate to 2025.9.1 and is still not working at first 2025.9.2 and 3
What's strange is that one of my esp32 boards ( esp32 std) running ModBus is updating, while my other 9 have the same cmake issue

---------------------------------------------------------
INFO Compiling app...
Processing home-assistant-voice-0952df (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
INFO Package configuration completed successfully
INFO Package configuration completed successfully
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50402.0 (5.4.2) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!

CMake Error at CMakeLists.txt:2 (include):
  include could not find requested file:

    /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

========================== [FAILED] Took 0.87 seconds ==========================

swoboda1337, 2025-10-02:
There was probably an error when you first installed the tools in the first build after reinstalling. 

It will take a long time to install, did you ever interrupt it?

swoboda1337, 2025-10-02:
After you reinstalled the builder did you ever interrupt the build?

SteinarSun, 2025-10-02:
Hello,
I haven't interrupted any compiling as far as I know, thought it was jus a issue with GitHub limitations.
Shoul I try removing EspHome builder and reinstall or is there any thing I need to upgrade/change ?

SteinarSun, 2025-10-02:
Ok, thanks for the info. Will wait for next release, the boards runs fine besides the trouble updating to latest firmware

kolodziejczyk123, 2025-10-03:
I updated to 2025.9.3, cleared everything and the problem is still active:

INFO ESPHome 2025.9.3
INFO Reading configuration /config/esphome/falownik.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing deye-inverter (board: esp32dev; framework: arduino; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip)
--------------------------------------------------------------------------------
Platform Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip
INFO Installing https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21-2/platform-espressif32.zip
Platform Manager: Error: Please read https://bit.ly/package-manager-ioerror
ERROR Error: Please read https://bit.ly/package-manager-ioerror
SSLError: Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 464, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1093, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 790, in connect
    sock_and_verified = _ssl_wrap_socket_and_match_hostname(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connection.py", line 969, in _ssl_wrap_socket_and_match_hostname
    ssl_sock = ssl_wrap_socket(
               ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/ssl_.py", line 480, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls, server_hostname)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/ssl_.py", line 524, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ssl.py", line 455, in wrap_socket
    return self.sslsocket_class._create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/ssl.py", line 1041, in _create
    self.do_handshake()
  File "/usr/local/lib/python3.12/ssl.py", line 1319, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'release-assets.githubusercontent.com'. (_ssl.c:1010)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 488, in _make_request
    raise new_e
urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'release-assets.githubusercontent.com'. (_ssl.c:1010)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/requests/adapters.py", line 644, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/connectionpool.py", line 841, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/urllib3/util/retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='release-assets.githubusercontent.com', port=443): Max retries exceeded with url: /github-production-release-asset/828921070/7a970c86-8f64-406c-b9b8-3dd3dba72d1d?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-10-03T18%3A42%3A01Z&rscd=attachment%3B+filename%3Dplatform-espressif32.zip&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-10-03T17%3A41%3A30Z&ske=2025-10-03T18%3A42%3A01Z&sks=b&skv=2018-11-09&sig=2cARjR183UAgx%2F%2F6XUcOozsar0tf7gnw8o6wxtLFTZ8%3D&jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1OTUxMzgyMCwibmJmIjoxNzU5NTEzNTIwLCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.JSH9seti9KmVgOYKaOfCcMbEs31i3_55qeWay4pRHW8&response-content-disposition=attachment%3B%20filename%3Dplatform-espressif32.zip&response-content-type=application%2Foctet-stream (Caused by SSLError(SSLCertVerificationError(1, "[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'release-assets.githubusercontent.com'. (_ssl.c:1010)")))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/usr/local/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
[... 62 lines cut ...]
[... cut ...] '. (_ssl.c:1010)")))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

kolodziejczyk123, 2025-10-04:
> You have a network problem.
> Can you access https://release-assets.githubusercontent.com/ from a
> computer? (it will be a 404)
> Do you have some sort of http proxy?

…

No, I don’t use any HTTP proxy. I’m on a home network (T-Mobile mobile
internet).

Error 404 Not Found

Not Found
Error 54113

Details: cache-iad-kjyo7100145-IAD 1759562227 1103899190
------------------------------

Varnish cache server

uspass, 2025-10-08:
Same here. Including:
```
Error 404 Not Found

Not Found
Error 54113

Details: cache-iad-kjyo7100145-IAD 1759904220 357708282

Varnish cache server
```
Reinstalling ESPHome Builder and restarting Home Assistant (HA) solved the problem.
If HA isn’t restarted, it behaves the same way.

On the other hand, I get the following warnings and an error (though the final result seems OK):
```
Warning: DEPRECATED: 'esptool.py' is deprecated. Please use 'esptool' instead. The '.py' suffix will be removed in a future major release.
Creating ESP32 image...
Successfully created ESP32 image.
Generating partitions .pioenvs/pump-command/partitions.bin
Generating an empty partition .pioenvs/pump-command/ota_data_initial.bin
Linking .pioenvs/pump-command/firmware.elf
usage: esp_idf_size [-h] [--format {table,text,tree,csv,json2,raw,dot}]
                    [--archives] [--archive-dependencies] [--dep-symbols]
                    [--dep-reverse] [--archive-details ARCHIVE_NAME] [--files]
                    [--diff MAP_FILE] [--no-abbrev] [--unify] [--show-unused]
                    [--show-unchanged] [--use-flash-size] [--lto] [-d]
                    [-o OUTPUT_FILE] [-s COLUMN] [-F PATTERN] [--sort-diff]
                    [--sort-reverse] [-q] [--no-color] [--force-terminal]
                    [--doc]
                    MAP_FILE
esp_idf_size: error: unrecognized arguments: --ng
Warning: esp-idf-size exited with code 2
RAM:   [=         ]  10.6% (used 34764 bytes from 327680 bytes)
Flash: [=====     ]  53.8% (used 986850 bytes from 1835008 bytes)
Building .pioenvs/pump-command/firmware.bin
Warning: DEPRECATED: 'esptool.py' is deprecated. Please use 'esptool' instead. The '.py' suffix will be removed in a future major release.
Creating ESP32 image...
Successfully created ESP32 image.
merge_factory_bin([".pioenvs/pump-command/firmware.bin"], [".pioenvs/pump-command/firmware.elf"])
Info: bootloader.bin not found - skipping
Info: partition-table.bin not found - skipping
Info: ota_data_initial.bin not found - skipping
Info: pump-command.bin not found - skipping
Using FLASH_EXTRA_IMAGES from PlatformIO environment
Merging binaries into /data/build/pump-command/.pioenvs/pump-command/firmware.factory.bin
Merging binaries with esptool
SHA digest in image updated.
Wrote 0x101070 bytes to file '/data/build/pump-command/.pioenvs/pump-command/firmware.factory.bin', ready to flash to offset 0x0.
Successfully created /data/build/pump-command/.pioenvs/pump-command/firmware.factory.bin
esp32_copy_ota_bin([".pioenvs/pump-command/firmware.bin"], [".pioenvs/cpump-command/firmware.elf"])
Copied firmware to /data/build/pump-command/.pioenvs/pump-command/firmware.ota.bin
======================== [SUCCESS] Took 390.94 seconds ========================
```

swoboda1337, 2025-10-16:
You can try latest and see if it help?
```
esp32:
  framework:
    version: latest
```

swoboda1337, 2025-10-16:
Did you try to do clean all (top right in the builder)

closed 2026-01-22 by github-actions[bot]