esphome/esphome #10956: Waveshare ESP32-P4 Smart 86 Box Development Board: wifi issues between P4 & C6 using esp32_hosted

DSew20, 2025-09-30:
### The problem

I’m trying to use ESPHome on an **ESP32-P4** host with an **ESP32-C6 co-processor over SDIO** via the esp32_hosted integration to provide Wi-Fi. The node boots, the SDIO link appears to initialize, but **Wi-Fi association is unstable and usually fails**. Typical behavior:

Shortly after starting Wi-Fi, the log prints ESP-Hosted SDIO errors like:
- H_SDIO_DRV: Dropping packet(s) from stream
- H_SDIO_DRV: Failed to push data to rx queue

Then the connection attempt **times out**, followed by:
- Disconnected ... reason='Association Leave'
- wifi: Restarting adapter

This **loops repeatedly**. In some runs I also see:
- assert failed: netif_add ... (netif already added) during Wi-Fi re-init
- occasional Guru Meditation / Instruction access fault during early setup

If I comment out the wifi: block, the device runs stably (no reboots/panics), which points to the Wi-Fi path (hosted/SDIO) as the trigger.

What I expected: the P4 should associate/authenticate via the C6 and obtain an IP (DHCP or static) and remain connected.

What happens instead: repeated association failures/timeouts, SDIO RX drops, and occasional asserts/panics when the adapter restarts.

### Which version of ESPHome has the issue?

2025.9.1 & 2025.9.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Component causing the issue

esp32_hosted, wifi, esp32

### YAML Config

```yaml
# base YAML #
esphome:
  name: p4-hosted-wifi-test
  friendly_name: P4 Hosted WiFi Test

esp32:
  board: esp32-p4-evboard
  framework:
    type: esp-idf

logger:
  level: DEBUG
  hardware_uart: UART0

api:

ota:
  - platform: esphome
    

# ESP32-C6 co-processor over SDIO
esp32_hosted:
  variant: ESP32C6
  reset_pin: GPIO54
  active_high: false
  clk_pin: GPIO18
  cmd_pin: GPIO19
  d0_pin: GPIO14
  d1_pin: GPIO15
[... 143 lines cut ...]
Other symptoms seen in some builds/runs:

- E ... esp_netif_lwip: dhcp client start failed

- esp_task_wdt_reset(...): TWDT was never initialized (warnings only)

- assert failed: netif_add ... (netif already added) during rapid adapter restarts

- occasional Guru Meditation (Instruction access fault) on Core 0/1 during setup

Environment details:

- ESP-IDF 5.4.x (platform 54.03.21)

- PSRAM enabled (200 MHz) and LDO configured; plenty of heap available

- Host: ESP32-P4; Co-proc: ESP32-C6 (ESP-Hosted firmware)

Hypothesis / request: The SDIO RX queue appears to overflow during association (handshake/scan), causing critical packet loss and a failed connect; subsequent rapid re-inits sometimes trigger netif_add asserts or panics. Guidance on:

- known fixes in ESPHome 2025.9.2+ or newer ESP-Hosted blobs,

- recommended SDIO buffer/queue tuning or SDIO clock adjustments for P4,

- exposing a YAML/IDF knob to increase Hosted RX queue / use PSRAM for Wi-Fi buffers, would be very helpful.

swoboda1337, 2025-09-30:
ESPHome has pinned the hosted component version to `2.0.11`: https://github.com/esphome/esphome/blob/dev/esphome/components/esp32_hosted/__init__.py#L97

Version loaded onto the C6 is probably not compatible. Do you know what version C6 is running? 

You can download the build for C6 here: https://components.espressif.com/components/espressif/esp_hosted/versions/2.0.11/readme?language=en

Can download archive and there is a slave folder and you can build the C6 firmware.

DSew20, 2025-09-30:
THANK YOU.

I spent days struggling, searching, testing, trying all variations possible and this fixed it.

closed 2025-10-01 by swoboda1337

tiimsvk, 2025-10-04:
Could you provide the binary files for flashing? Thanks.